foxglove/studio #1112: ROS1 connection cannot receive data if node hostnames are not resolvable

rgov, 2021-05-29:
When I connect to my ROS master (by IP address), Studio seems to connect OK and get a list of topics that are being published. But if I try to subscribe to one of these inside a panel, this happens:

    XmlRpcClient.ts?3553:34 POST http://myrobot:41045/ net::ERR_NAME_NOT_RESOLVED
    RosNode.ts?c420:728 [ROS1] subscribing to /tag_detections_image at http://myrobot:41045/ failed (Error: requestTopic("/tag_detections_image") from http://myrobot:41045/ failed. err=TypeError: Failed to fetch), this connection will be dropped

What I'm guessing is happening is that the ROS master (on which the `/tag_detections_image` topic publisher lives) knows itself by the hostname `myrobot` but the computer running Studio cannot resolve that name. If I manually add `myrobot` to the hosts file so that it is resolvable, then panels are able to subscribe to topics as they'd like.

I'm not sure if it's up to Studio to fix this, but some diagnostics would be better than appearing to do nothing.

jhurliman, 2021-05-29:
This is a ROS1 configuration issue (see the ROS_HOSTNAME environment variable documentation). rosmaster will happily publish whatever name nodes advertise themselves as, so they should choose a hostname all other nodes (including Studio) will be able to resolve. Let’s keep this ticket open to improve the error reporting around this common issue.

defunctzombie, 2021-06-01:
I ran into similar issues when trying to diagnose why I wasn't getting messages that other nodes should be sending. For ROS connections we should find a way to surface what studio is connected to vs what it thinks it should be connected to. We could do this via the player problems in the basic case and in the best case maybe some UI that would show you the established connections / errored connections.

defunctzombie, 2022-06-08:
Closing this as the question was answered.