foxglove/studio #1132: display recommendation to switch to Chome

wkalt, 2021-06-01:
In the web build at studio.foxglove.dev, a firefox user will see this on page load (before attempting to load a bag):

[image]

Firefox support isn't a requirement for us today, but we should display a message to try the site with Chrome, rather than the error.

jhurliman, 2021-06-02:
The error is:

```
Error: Chart requires browsers with offscreen canvas support
    <anonymous> index.tsx:144
```

amacneil, 2021-06-02:
I think we should take down the web build for now since it is not officially supported. It can stay up on a non public domain.

closed 2021-06-02 by amacneil