foxglove/studio #1158: Display notification when plot downsamples data

defunctzombie, 2021-06-03:
The plot panel downsamples data to improve performance for large data sets. For unaware users the downsampling appears as data loss or incomplete data and they wonder if something is broken.

We should display a notice indicating data has been downsampled and that zooming in will show higher fidelity data.

defunctzombie, 2021-06-10:
https://github.com/foxglove/studio/pull/1188 dialed down the aggressiveness of the downsampling and made the data points a bit smaller to accommodate the denser data.

Another idea instead of a notification is to show lines only (no data points) when the data is downsampled as a visual hint.

amacneil, 2021-07-19:
I'm not convinced that we need this. Every metrics/charting platform I have used downsamples data over a large time window.

closed 2021-07-19 by amacneil