foxglove/studio #1176: Support for dynamic_reconfigure in Parameter panel

rgov, 2021-06-08:
The Parameter panel does not appear to support [dynamic_reconfigure](http://wiki.ros.org/dynamic_reconfigure). This would be useful for tuning parameters for live systems.

The Parameter panel does show params that are "managed" by dynamic_reconfigure because it reflects them to the parameter server. But, when they are updated in Studio, it does not trigger the reconfigure callback to allow the node to react to the new value.

Also, dynamic_reconfigure gives information like documentation strings and min and max ranges which might be useful for presenting a UI as `rqt_reconfigure` does.

rgov, 2021-06-08:
I think you would need to do two things:

1. Find topics of type [dynamic_reconfigure/ParamDescription](http://docs.ros.org/en/noetic/api/dynamic_reconfigure/html/msg/ParamDescription.html) and interpret the parameter descriptions to build a UI

2. Find services that match [this interface](http://docs.ros.org/en/noetic/api/dynamic_reconfigure/html/srv/Reconfigure.html), usually `/<node>/set_parameters`, and invoke the service whenever the UI updates, and use its return value to override what you display in the UI

3. Subscribe to `/<node>/parameter_updates` as well to update the UI.

rgov, 2021-06-08:
Related to #790.

DevnathNair, 2021-06-17:
This would be amazing. Would make it all the more functional.

sink0918, 2021-08-21:
Please do add this feature to Studio, it is very useful for me too!

jhurliman, 2022-02-17:
I just bumped into this setting up the `realsense2_camera` node which makes heavy use of dynamic_reconfigure for live tuning of the camera. Being able to tune in studio without launching rqt_reconfigure would be lovely.