foxglove/studio #1222: Add memory and fps stats to application

defunctzombie, 2021-06-14:
We should track application wide memory use (at least in dev mode) so we can keep an eye on potential memory issues. We could do this with a panel or maybe with a status bar.

We should track delays _between_ requestAnimationFrame calls. Since we don't have a single "render" loop we can at least determine if other things are stalling the render thread by looking at the delay between raf calls.

# Inspiration and references
https://github.com/mrdoob/stats.js/
https://github.com/electron/electron/issues/4173

jhurliman, 2022-01-07:
We still need this!

defunctzombie, 2022-07-08:
Amusingly I'd forgotten about this issue but did make a POC: https://github.com/foxglove/studio/pull/3748

jtbandes, 2022-12-13:
Amusingly you forgot about this issue again!

closed 2022-12-13 by jtbandes