foxglove/studio #1240: Feature request: Add publish support to extension api

defunctzombie, 2021-06-15:
To create teleop panels (i.e. turtlesim teleop) we need publish support in the extensions api.

Some concerns to think through:
- Since initPanel could be called multiple times what happens to registered publishers? Does the extension author need to clean them up?
- Do we require the message definition passed in explicitly or support datatype string and definitions? I'm tempted to say we should require explicit definitions since datatype strings are a rosism?
- Does the api for publishing return a promise until the message is published or is it fire and forget?

jhurliman, 2021-06-21:
>  since datatype strings are a rosism

Our explicit message definition types are ROSisms as well. A string identifier for datatypes would cover many serialization formats such as protobufs, thrift, etc while the explicit definitions would not.

jhurliman, 2021-06-21:
> Does the api for publishing return a promise until the message is published or is it fire and forget?

I think this should be fire and forget since the concept of "published" or "not published" is fuzzy. There can be multiple subscribers with separate states of having received or not received each message and the transport might support acknowledgment (i.e. UDP).

closed 2021-08-13 by defunctzombie