foxglove/studio #1258: Ability to visualize two separate TFs at the same time

jtbandes, 2021-06-17:
As reported in https://github.com/cruise-automation/webviz/issues/649 (cc @theonlynick0430). Users would like to visualize 2 separate robot poses at the same time, and since the 3D panel doesn't support custom 3D models, they're using TF frame visualization as a workaround.

However, the current 3D panel 2-bag/2-column mode doesn't support visualizing both bags' TF frames:

https://github.com/foxglove/studio/blob/368e36cb337191afc54d65de1e3c91ae86142e69/packages/studio-base/src/panels/ThreeDimensionalViz/TopicTree/renderNamespaceNodes.tsx#L248-L251

We could simply implement that code path, but some alternatives would be:
- If we implement #782 then that takes care of some of the motivation behind this request.
- We could allow the user to select which TF topic(s) are processed by the 3D panel, so we don't need the 2-column interface for dual bags. Maybe something like the TopicToRenderMenu
[image]
- We could treat TF more like a regular topic with various frames as namespaces, and then it would use the regular 2-column support like other topics (although Marker namespaces don't allow arbitrary nesting, while the TF tree does).

amacneil, 2021-12-15:
@jtbandes I'm not sure what the specific request in this issue is now, but is it still relevant given our recent support for markers, transform handling, lack of dual bag support, etc?

jtbandes, 2021-12-15:
Given we removed dual bag support I guess this is not relevant. If we ever add it back it would become relevant. 🤷

closed 2021-12-15 by amacneil