foxglove/studio #1261: Node Playground should support customizing datatype name

jtbandes, 2021-06-17:
Split from #726 

The common datatypes introduced in #1038 should be exposed to Node Playground code so that it can use common ROS datatypes even if they are not in the bag/data source.

There is currently a hack specifically to allow publishing `visualization_msgs/MarkerArray`: 

https://github.com/foxglove/studio/blob/860b4df2cad85a39bca022749f80a0a48c6a8796/app/players/UserNodePlayer/nodeTransformerWorker/typescript/ast.ts#L322-L323

jtbandes, 2021-06-23:
I just discovered there is some logic that translates `a_msgs__B` to `a_msgs/B` in the transformation code: https://github.com/foxglove/studio/blob/ffe3c722d645ede0e7f81d69002261bd59762c60/packages/studio-base/src/players/UserNodePlayer/nodeTransformerWorker/transformer.ts#L365-L368

bunnyc1986, 2021-11-23:
I found the fix it to add `basicDatatypes` into the `generateRosLib`'s `datatypes` Map

https://github.com/foxglove/studio/blob/main/packages/studio-base/src/players/UserNodePlayer/index.ts#L578-L581

to 
```
    const rosLib: string = await transformWorker.send("generateRosLib", {
      topics,
      datatypes: new Map([...basicDatatypes, ...datatypes]),
    });
```

PR #2289 (2021-12-07): Add basicDatatypes to UserNodePlayer roslib
**User-Facing Changes**
User node authors can output messages of any basic datatype even if the datatype is not present in the datasource.

**Description**
User nodes are only able to output known datatypes. Prior to this change, known datatypes would be read from the current data source. This limited the types of messages you can output from user nodes to the datatypes in the data source.
    
This change adds the basic datatypes to the known datatypes for user nodes. This allows a user node author to output any of the basic datatypes even if not pr [... cut]
commit b26b5b98 (2021-12-07): Add basicDatatypes to UserNodePlayer roslib (#2289)

User nodes are only able to output known datatypes. Prior to this change, known datatypes would be read from the current data source. This limited the types of messages you can output from user nodes to the datatypes in the data source.

This change adds the basic datatypes to the known datatypes for user nodes. This allows a user node author to output any of the basic datatypes even if not present in the data source.

Fixes: #1261
closed 2021-12-07 by defunctzombie