foxglove/studio #1273: Support bgra8 images

jtbandes, 2021-06-18:
The Image panel currently supports rgb8/bgr8, but not rgba8/bgra8 images. Some relevant references:

* http://docs.ros.org/en/jade/api/sensor_msgs/html/namespacesensor__msgs_1_1image__encodings.html
* http://wiki.ros.org/cv_bridge/Tutorials/ConvertingBetweenROSImagesAndOpenCVImagesPython

WaelKarkoub, 2022-04-18:
Hello, will you guys be supporting bgra8 in the near future? Having the same issues

jtbandes, 2022-04-18:
This isn't currently on our roadmap, but we would be happy to review a pull request if anyone is interested in contributing. It should be relatively straightforward to add/modify the code here:

https://github.com/foxglove/studio/blob/main/packages/studio-base/src/panels/ImageView/lib/decodings.ts

https://github.com/foxglove/studio/blob/735fff89303a526ea10af595c3cdf0725fab46ba/packages/studio-base/src/panels/ImageView/lib/renderImage.ts#L139-L145

norbertmarko, 2022-06-15:
Hi! 
I've seen that you have added `rgba8` support and thank you for this! Unfortunately, our `.bag` files contain Images in `bgra8` format. Is it possible to add support for this type with a little alteration of the new commit? We would be very grateful for it!

defunctzombie, 2022-06-15:
@norbertmarko We'd be happy to review a PR if you want to implement bgra8 support. It should be easy to add by copying the `rgba8` decode function and changing the fields around.

jhurliman, 2022-07-08:
Fixed in https://github.com/foxglove/studio/pull/3796

closed 2022-07-08 by jhurliman