foxglove/studio #1291: /clock from outside sources breaks rendering

alluring-mushroom, 2021-06-23:
**Description**
I am publishing a custom time on the /clock topic, and setting my nodes to `use_sim_time:=true`. 
When I use this clock topic, the point cloud in the 3D panel "flickers" it displays the latest message, the stops displaying it (almost immediately) then displays the next message when it arrives.

I am generating the topic data in ROS2, then bridging it across, however I am able to reproduce the bug via ros1 bag file.

When I run my nodes in real-time, or run the bag file with the `/clock` topic muted, the issue vanishes.

When I drag and drop either bag file (one with `/clock` and one without), Foxglove plays both back perfectly.

- Version: 0.12.0
- OS: Ubuntu 20.04

**Steps To Reproduce**

[example.zip](https://github.com/foxglove/studio/files/6698421/example.zip)

1. Play bag file `simclock_2021-06-23-11-15-43.bag` via rosbag
2. open a 3d panel in Foxglove, showing the topic /`points`
3. you will see the points "flickering"
4. play either `realtime_2021-06-23-11-11-39.bag` or the simclock bag file excluding `/clock`
5. the issue no longer exists

**Expected Behavior**
Pointcloud should render correctly regardless of `/clock` source

**Actual Behavior**
Pointcloud rendering breaks

defunctzombie, 2021-06-23:
@alluring-mushroom I played back both bag files in example.zip and did not see any flickering in the 3d panel. I used version 0.12.0 but it was on MacOS.

I see in your example.zip you've recorded a video. That is a recording from using a live ros connection rather than the bag playback. I want to confirm that you are seeing flickering in the bag files you uploaded in the zip file - as I am not.

How often and what does your fake /clock source generate? I'm thinking this issue is specific to the live ros1 connection and need to come up with a way to have a fake clock source and produce some point cloud data timed to this source similar to how you were doing with your live connection.

alluring-mushroom, 2021-06-23:
@defunctzombie 
Hey, thanks for looking into it.

Indeed, there is no issue when dropping the bag file into Foxglove, only when using either rosbag to play back the file or running live nodes. So far any 'external' clock

The `/clock` source generates Clock messages, and currently every 100ms

alluring-mushroom, 2021-06-23:
I've check what effect changing this has (10ms), and the result is that flickering is intermittent. The 3D panel will go 10s without flickering, then the next 5s will flash multiple times

defunctzombie, 2021-06-23:
Can confirm I am able to `rosbag play` the simclock bag (with and without the --clock flag to rosbag) and see flickering in the 3d panel when connected using the ROS connection in Studio.

siddancha, 2021-10-31:
TLDR; I have the same issue, and I am using ROS-1 with `use_sim_time := true`.

---

When running the desktop app with TCP connections (when running the rosmaster), 3D panels flicker. See attached video, where both `sensor_msgs/PointCloud2` and `nav_msgs/OccupancyGrid` topics flicker.

This does not seem to happen (or happens extremely rarely) when playing a recorded bag file. So I'm not sure how much help the attached bag file will be.

- Version: 0.20.1
- OS: Ubuntu 20.04

**Steps To Reproduce**

Zipped bag file: [data.bag.zip](https://github.com/foxglove/studio/files/7448137/data.bag.zip)
 
**Expected Behavior**

3D panels shouldn't flicker.

**Actual Behavior**

https://user-images.githubusercontent.com/6638091/139568830-ca3ba822-eeb1-4daa-8aad-15ba5d33cab8.mp4

jhurliman, 2021-12-08:
Fixed in https://github.com/foxglove/studio/pull/2322

closed 2021-12-08 by jhurliman