foxglove/studio #1299: Move Node Playground editor into sidebar

defunctzombie, 2021-06-24:
Rather than keeping the node playground hidden in a panel (and confusingly allowing you to add multiple node playground editors?) move it to the sidebar where a user is more likely to discover it.

Right now node code is saved with the layout which also means you can't easily use the same node in another layout. Separating that behavior has pros/cons.

Cons
- sharing a layout json file is insufficient to share node code

Pro
- nodes re-usable across layouts
- nodes shared with your org

amacneil, 2021-06-24:
We should definitely separate NP from the layout. A few points to consider:

**Sharing**: if we remove NP from the layout we need a replacement solution for people to share them with their org.
- Saving NP node within your org might be nice.
- Could allow linking github account and committing there (Looker uses this workflow, it's really nice imo). This could be combined with the concept of org shared nodes (i.e. the github config is global to your org, everyone commits to the same repo).
- Could allow bundling NP nodes up into an extension. It definitely feels like NP could just be a type of extension, and we happen to have a live editor available (maybe the live editor could let you edit other extension types too? or maybe we do away with the inline editor altogether and make filesystem watching work properly, although this increases the barrier to entry for folks not familiar with JS/TS dev)

**UI/UX**:
One downside with the sidebar specifically is that it might not be wide enough to be useful. It's possible to resize, but then that seems a little weird once I switch to "Add Panel" or another "normal" sidebar and it's still taking up half my screen. Some other ideas to consider:
- Some concept of top level tabs, like VScode? Ability to open a layout in each tab, or have NP in one of the tabs?
- Some concept of "fullscreening" a sidebar (maybe by double-clicking)? This would allow NP to take over the full screen, but once you are in the final edit/tweaking stages you could switch it back to a "skinny" sidebar so that you can edit and see changes in real time.

**Name**: We also need a better name for Node Playground. I would get away from the concept of nodes entirely

defunctzombie, 2021-06-24:
made https://github.com/foxglove/studio/pull/1300 so we could explore some of the UI/UX implications.

This immediately exposed an annoyance around selecting a connection (or using any other sidebar item) replacing my editor.

defunctzombie, 2021-06-24:
I like `Some concept of top level tabs, like VScode? Ability to open a layout in each tab, or have NP in one of the tabs?` but a nice feature of node playground is the ability to make edits and observe the change in panels without having to toggle between tabs etc.

We could consider a separate "editor" window - that allows the user to arrange the windows using their window manager - we lose some control over placement tho.

amacneil, 2021-06-24:
> ability to make edits and observe the change in panels without having to toggle between tabs

Agreed. One possible solution is to allow "split screen" with a tab in each half (again similar to vscode). But this might start getting a bit weird/confusing with our existing concept of panels. Still, might be worth exploring the idea.

closed 2022-01-22 by github-actions[bot]

amacneil, 2022-01-26:
In case anyone searching comes across this feature:

We still have plans for improving Node Playground in the future, but it's not a high priority and there's not much point keeping non-specific issues open. If you have use cases to share, please let us know!

amacneil, 2022-02-14:
Concrete thought from discussion today:

Node playground should exist as a sidebar menu item, similar to layouts. It can have a list of nodes, with a plus icon to create new nodes.

When you edit a node (maybe right click->edit), we can add the editor panel to your current layout. That way, we avoid the current UX mess of sidebars inside panels, but we can still let people use the editor alongside visualizing the result of their edits.