foxglove/studio #1332: Add _blank target to various anchor tags

defunctzombie, 2021-06-29:
In https://github.com/foxglove/studio/pull/1329 I removed the _blank target attribute for anchor tags since they are no longer necessary for studio. They are still necessary for web to avoid the user replacing the current studio page with the target page.

jtbandes, 2022-01-03:
I know we've filed some tickets for specific instances of this issue, e.g. https://github.com/foxglove/studio/issues/2438

Is there any general solution or lint rule to be made here? Any reason to keep this ticket open?

defunctzombie, 2022-01-03:
Closing. We fixed this for the desktop for all links. For web we need to be more diligent as 2438 notes.

closed 2022-01-03 by defunctzombie

jtbandes, 2022-01-03:
I think this ticket was about web though...

defunctzombie, 2022-01-03:
Yea - I think its better for us to fix this when we find it. Maybe a lint rule or component. We can keep 2438 open as our more active ticket since it shows a specific case within the app.