foxglove/studio #1345: URDFViewer sentry error when there is no joint state name

defunctzombie, 2021-06-30:
**Description**

```
TypeError: Cannot read property 'forEach' of undefined
```

[image]

- Version: 0.12.0
- OS: Ubuntu 18.04

https://sentry.io/organizations/foxglove/issues/2484956494/?project=5649767&referrer=slack

jtbandes, 2021-06-30:
Any repro steps? How is there no name? Was this using node playground?

Lots of panels assume messages conform to a certain shape based on the datatype name. Adding `?.` everywhere seems like overkill and we’d need to change all our types to be DeepPartial to enforce it. So maybe we need more of an upfront validation step? (But running this on each message could be expensive)

defunctzombie, 2021-06-30:
I've not tried to repro - only hoisted the sentry error to a github issue. Might be doable to reproduce with node playground. Either we need the panels to be more defensive or a better mechanism to catch the errors within the panel to prevent them bubbling to sentry. Tho if we do catch the error in the panel it will be rather opaque for the user what action to take unless we do perform some sort of validations on the messages.

jtbandes, 2021-08-13:
Closing since the Sentry issue was resolved due to inactivity, and the error doesn't seem particularly actionable unless we know more about how to reproduce it. Only 1 instance was ever recorded in Sentry.

closed 2021-08-13 by jtbandes