foxglove/studio #1351: Fail to connect to ROS with rosbridge websocket without login

Oasis-Guo, 2021-07-01:
Hi, I fail to use this tool for the login issue.

I use Frp tool to let robots that using inside networks or doesn't have a publich ip address , can access from outside network. With this method, I  success to make the rosbridge works, and access it using any network. But for the safety issue, I set a password for the websocket. For now, I can use webviz to debug and what I need just need to login websocket every time. After login, I can start to use webvie. But I don't know how to login first using foxglove, and if I don't do the login first, then It is not accessable to use the rosbridge connection.

below is how I login to rosbridge websocket with Chrome Browser:

[image]

defunctzombie, 2021-07-01:
Have you tried using a websocket url of the following form: `ws://username:password@host:port` ? While not idea that you need to list the username and password in the url this might work depending on how the webserver that is requesting authorization to access the url is setup.

Oasis-Guo, 2021-07-02:
Nice!!!!! It is working right now with ws://username:password@host:port 

Thanks a lot

closed 2021-07-02 by jtbandes