foxglove/studio #1414: Provide an easy way for users to get in-app help

esthersweon, 2021-07-07:
Additional "Help" tab in the sidebar witH:
- [Chat window](https://papercups.io) or comment form for in-app feedback
- Link to the app's [docs](https://foxglove.dev/docs)

If we go with a comment form, we can automatically open a GitHub issue based on the input.

**Benefits:**
- **Get more user feedback** – Get in-app feedback from users in the middle of a frustrating problem; reach users who don't want to join another Slack channel or find the Github repo to open an issue
- **Get regular user feedback on both released & WIP features** – Include user surveys to get feedback on newly released or WIP features as a way to validate what we're building (e.g. "how do you feel about team layouts? why?", "how would you rate Foxglove Studio? why?") 

cc: @2metres

amacneil, 2021-07-07:
Quite a few ideas in this ticket - is there an mvp? 

Have we had anyone contacted us via papercups on the website? 

Worth considering that in app chat support might drive people away from slack (public venue) to private conversations, which has less value to the community.

amacneil, 2021-11-15:
Most of this was implemented in https://github.com/foxglove/studio/pull/2140, so closing this issue.

@esthersweon is planning to make ongoing improvements to help sidebar.

closed 2021-11-15 by amacneil