foxglove/studio #1422: Add webkit support for web build

esthersweon, 2021-07-07:
We need webkit support for web build, since most users use iOS (where Safari and all other web browsers all require webkit).

This will unlock mobile support to some extent; more work needs to be done to optimize UI for touch devices.

defunctzombie, 2021-07-07:
Any notion of what doesn't work today? Breaking this down into smaller "this doesn't work" parts would be the first step in determining the work involved in supporting. I think generally its going to be tricky with more advanced uses of browser tech that typically appears in Chrome (i.e. filesystem support, maybe some bigint things?). I'm not sure this is something we will prioritize short term unless its a few small tweaks.

esthersweon, 2021-07-07:
@defunctzombie – can go through and find the "this doesn't work" parts once we have a production web build going (i.e. after login flow is added).

esthersweon, 2021-07-07:
@defunctzombie – from @jhurliman: "i think the only real thing holding us back from iOS right now is a lack of polyfill for dataview bigint getters/setters"

jtbandes, 2021-07-07:
I started working on that in https://github.com/foxglove/studio/commit/76e51ededb9c0d929b3c9a32abe3c5f984d3ef6b but it fell by the wayside; could pick it up again

defunctzombie, 2021-07-07:
https://caniuse.com/mdn-javascript_builtins_dataview_getbigint64
https://caniuse.com/mdn-api_window_showopenfilepicker

jtbandes, 2021-12-23:
Closing this issue as it is too broad (and includes a questionable premise: "most users use iOS").

The latest version of Safari supports BigInt64Array & DataView.getBigInt64.

Opened https://github.com/foxglove/studio/issues/2505 for showOpenFilePicker.

amacneil, 2021-12-23:
@jtbandes besides individual features that need to land, I think it's worth having some point in time where we remove the warning banner and claim safari is supported. Is that time now? Or should we keep this ticket open to track that?

We can edit conjecture out of the ticket description...

jtbandes, 2021-12-24:
Not sure what else may be missing. Probably someone needs to do a runthrough of all the features and make sure things are generally working. I guess if we want to claim Safari/Firefox are supported we probably shouldn't throw up a "performance may be degraded" toast either...

jtbandes, 2021-12-24:
What would you envision us doing to make sure we maintain compatibility in the future (since most of our dev work happens in Chrome)?

amacneil, 2021-12-24:
> What would you envision us doing to make sure we maintain compatibility in the future (since most of our dev work happens in Chrome)?

This is probably the real question. We probably need a Chromatic like service that can handle cross-browser testing.

Sounds like Chromatic already supports FF: https://www.chromatic.com/docs/browsers

rgov, 2022-04-21:
I think you are missing OffscreenCanvas, no? https://caniuse.com/offscreencanvas