foxglove/studio #1426: Convert to function component

defunctzombie, 2021-07-08:
Some components remain using classes. They should be assimilated into the function component borg.

- [ ] packages/studio-base/src/components/Autocomplete.tsx
- [ ] packages/studio-base/src/components/ButtonBase.tsx
- [ ] packages/studio-base/src/components/Dropdown/index.tsx
- [ ] packages/studio-base/src/components/KeyListener.tsx
- [ ] packages/studio-base/src/components/Menu/Menu.tsx
- [ ] packages/studio-base/src/components/Menu/SubMenu.tsx
- [ ] packages/studio-base/src/components/MessagePathSyntax/MessagePathInput.tsx
- [ ] packages/studio-base/src/components/NotificationDisplay.tsx
- [ ] packages/studio-base/src/components/Select/Option.tsx
- [ ] packages/studio-base/src/components/Select/Select.tsx
- [ ] packages/studio-base/src/components/Slider.tsx
- [ ] packages/studio-base/src/panels/ThreeDimensionalViz/DrawingTools/MeasuringTool.ts
- [ ] packages/studio-base/src/panels/diagnostics/DiagnosticStatus.tsx
- [ ] packages/studio-base/src/panels/diagnostics/DiagnosticSummary.tsx

jtbandes, 2021-07-08:
Several of these should be removed and replaced with FluentUI:
- #703 
- #705 
- #702

defunctzombie, 2021-12-20:
Closing this out - The remaining items of interest have open PRs and we will track those individually.

closed 2021-12-20 by defunctzombie