foxglove/studio #1482: Document per-panel message schema and datatypes

defunctzombie, 2021-07-16:
Several panels have functionality which depend on messages conforming to a specific schema and datatype. For example, the image panel supports messages conforming to the [ImageMarker](http://docs.ros.org/en/melodic/api/visualization_msgs/html/msg/ImageMarker.html) datatype.

Rather than guessing the datatypes through trial-and-error or having to read the code, we should document our supported datatypes on the per-panel docs. We should assign foxglove datatype names rather than directing users to "read the ROS messages and types" since our goals are to be framework agnostic and support fields not present in ROS messages.

The image panel is another example of extending the supported fields beyond the ROS datatype. The image panel supports a "text" field within the image marker message (https://github.com/foxglove/studio/blob/main/packages/studio-base/src/panels/ImageView/renderImage.ts#L384).

defunctzombie, 2021-11-11:
https://github.com/foxglove/website/pull/226

amacneil, 2022-01-18:
related follow up work
- create .proto and JSON schemas for our foxglove messages
- rename https://github.com/foxglove/ros-message-schemas and add to that repo
- decide whether to keep the ROS .proto + JSON schema definitions around, or just remove them and prioritize foxglove schemas

amacneil, 2022-01-24:
Probably important to document on https://foxglove.dev/docs/messages/location-fix type pages that the data type name is important. 

Technically that is true for ROS 1 + ROS 2 as well, but those users would not normally be renaming their message types etc. However for custom formats they are going to be using our foxglove-websocket server so it's important that they know they need to send the correct datatype name.