foxglove/studio #1493: Extensions on web should prompt to use desktop version

amacneil, 2021-07-19:
**Description**
Extensions do not currently work on web, and fail with a useless error message.

We should keep the extension sidebar button, but disable the controls and prompt people to download the desktop build for extension support.

defunctzombie, 2021-07-20:
Initial thoughts on approaches:

- For extensions we could replace the sidebar component with one that directs user to download
- For extensions we could avoid providing an extension loader and treat the absence of a loader as a call to action - or less pretty - have the loader throw a more helpful error message saying "extensions are only available in the desktop version" or some such
- For connections - I've wanted to extract these out of player manager and accept an array of them into the App. The tricky bit here is around any UI the connection needs to expose.
- Pass a list of "capabilities" or "features" to the App and expose this to our various UI elements - the elements would see their feature missing and display "not available in this version" or something similar. I'm less fond of this approach since now various elements have multiple code paths.

amacneil, 2021-07-20:
For extensions it would be nice if we can keep the extension catalog there, but just disable the 'install' button and explain the need to download, or even just keep the install button there and have it trigger the modal instead of actually installing.

One other thing to keep in mind is that someone customizing studio-base might prefer to just disable the functionality altogether. But for our own web build we prefer these download prompts.

defunctzombie, 2021-07-20:
> For extensions it would be nice if we can keep the extension catalog there, but just disable the 'install' button and explain the need to download, or even just keep the install button there and have it trigger the modal instead of actually installing.

Unfortunately we cannot in the current form because we read the marketplace list from a json file in a github repo and CORS blocks that.

jtbandes, 2021-07-20:
> Unfortunately we cannot in the current form because we read the marketplace list from a json file in a github repo and CORS blocks that.

It sounds like GitHub Pages has proper CORS headers so if we enable Pages we might be able to fetch from there in a browser.

defunctzombie, 2021-07-20:
Related to the ROS1 and Velodyne requests: https://github.com/foxglove/studio/issues/776