foxglove/studio #1500: Bag causes app to freeze

amacneil, 2021-07-19:
**Description**
App freezes or crashes when loading this bag.

- Version: 0.13.1
- OS: Linux

**Steps To Reproduce**
1. Download and open https://storage.googleapis.com/cartographer-public-data/bags/backpack_3d/with_intensities/b3-2016-04-05-14-14-00.bag (it's quite large)
2. Add 3D panel and enable `horizontal_laser_3d` topic
3. When bag is about 15% finished parsing, the whole app crashes or freezes (both have happened to me at different times)

**Expected Behavior**
No crash

**Actual Behavior**
Crash :(

amacneil, 2021-07-19:
Can someone try to repro this on mac?

defunctzombie, 2021-07-19:
Is your "unlimited cache" app setting on or off? If you have that setting on - the app crashes because it tries to load the entire bag into memory and then OOMs.

amacneil, 2021-07-20:
I have "unlimited cache" disabled (side note - seems like a bad idea for a setting?).

When I load that bag, I see the process grow in `htop` until it reaches just under 2gb physical memory before the renderer process crashes.

defunctzombie, 2021-07-21:
Lets pair on this and figure out what is happening and how we can prevent it from crashing the app.

amacneil, 2021-08-19:
Still happening on v0.15.0 and latest main

closed 2021-08-31 by amacneil