foxglove/studio #1508: `?demo` URL param to restore welcome layout + demo connection

amacneil, 2021-07-20:
We want to link to studio.foxglove.dev from our homepage. We need a way to force it to re-load the demo layout and demo bag.

The most obvious solution would be a `?demo` URL parameter. It would also be nice to allow loading any arbitrary remote bag via a URL param, but we would still need the `?demo` option since we need this to restore the welcome layout too.

defunctzombie, 2021-07-20:
Instead of a `?demo` param - I think we should remove persistence from the web deployment. Every time you open the page it should be a clean instance which will open the demo by default.

amacneil, 2021-07-20:
Unless you pair that with keeping layout and connection state in the URL, that sounds incredibly annoying if hitting refresh changes my current layout and connection.

defunctzombie, 2021-07-20:
> Unless you pair that with keeping layout and connection state in the URL, that sounds incredibly annoying if hitting refresh changes my current layout and connection.

Yes - its a demo page - I think for a quick turnaround it should be the way to go - and later we can think about whether we do nicer things if you are logged in.

amacneil, 2021-07-20:
Sounds ok if it's the fastest solution for now, although seems equally easy to just look for `?demo` and leave the existing functionality as is?

It's not a demo any more, it's the real thing.

defunctzombie, 2021-07-20:
> Sounds ok if it's the fastest solution for now, although seems equally easy to just look for `?demo` and leave the existing functionality as is?

We'll see - I am very much against doing url/query handling and localstorage directly in studio-base. So its not so trivial to "just look for demo" but we also need to expose how to invoke demo out to the desktop/web platforms. I think we have something in place for this already since we have that help menu item on desktop that launches the welcome layout. Maybe something for @jtbandes to comment on since he wired that up.

jtbandes, 2021-07-20:
The `<Workspace/>` is what currently shows the demo layout on first run. We could add some kind of hook/prop/context for the web build to configure this if we want:

https://github.com/foxglove/studio/blob/ae18633658b9b216bc67134b1955829608e76373/packages/studio-base/src/Workspace.tsx#L270-L281

Note that currently `openWelcomeLayout()` always _saves the welcome layout as a new layout_ so it could get annoying if you refresh a bunch of times and your layout list explodes with demo layouts.

defunctzombie, 2021-07-20:
@jtbandes What would it look like to avoid saving the welcome layout?

jtbandes, 2021-07-20:
Depends how you want to handle it – for example we could list the layouts and see if any has a matching name, and if so just switch to that one.

If we want to avoid saving as you edit, would need to change around the CurrentLayoutProvider: https://github.com/foxglove/studio/blob/980c944c6f7c5251761dae5737744faaf4479293/packages/studio-base/src/providers/CurrentLayoutProvider/index.tsx#L106-L119

PR #1518 (2021-07-21): web: add ?demo url param to force loading welcome layout
**User-Facing Changes**
None (more for internal/marketing use)

**Description**
Fixes #1508 

If the `?demo` flag is present in the url, always load the welcome layout on first mount. Also do a `history.replaceState` to remove the demo flag.

(Note that this still means creating a _new_ layout each time.)
commit 005ced78 (2021-07-21): web: add ?demo url param to force loading welcome layout (#1518)

**User-Facing Changes**
None (more for internal/marketing use)

**Description**
Fixes #1508 

If the `?demo` flag is present in the url, always load the welcome layout on first mount. Also do a `history.replaceState` to remove the demo flag.

(Note that this still means creating a _new_ layout each time.)
closed 2021-07-21 by jtbandes