foxglove/studio #1511: Meta: Improve Cross-Browser Compatibility

amacneil, 2021-07-20:
This is a meta-issue tracking work happening to improve cross-browser compatibility. Currently, we only support Chrome (v76+?).

- [ ] Edge: I don't believe anyone has tested, but since it uses Blink it might work?
- [ ] Safari, and all browsers on iOS/iPadOS (which are forced to use Webkit) (https://github.com/foxglove/studio/issues/1422)
- [ ] Firefox

ic, 2022-01-17:
I would like to just share the latest container to this day does not work at all on Firefox. It has become such a minor browser that it may not be relevant, but main one here.

The error trace is little interesting, blocked at initialisation.

[image]

Also, I am on an beta version of the Firefox (97.0b2 (64-bit)), not always stable, yet everything else seems to work as expected.

Good luck with all the tasks ahead (beyond browser support!)

amacneil, 2022-01-17:
Yes, we don't support Firefox yet, since it's lacking a few features and we have not yet built workarounds (also, we have not yet set up cross-browser testing which is a pre-requisite to proper support).

However, the error you're seeing is weird. It looks like maybe related to an extension you have installed? I can visit [studio.foxglove.dev](https://studio.foxglove.dev) in FF (Ubuntu 20.04 ARM) just fine. Does that only happen in the container for you or does it also happen on [studio.foxglove.dev](https://studio.foxglove.dev)?

jtbandes, 2022-01-18:
Per instructions in the readme, I just tried running `docker run -it --rm -p "8080:8080" ghcr.io/foxglove/studio:latest` and visiting localhost:8080 in Firefox 96.0.1, and it works for me. What data source are you loading into Studio?

ic, 2022-02-09:
> We don't support Firefox yet, since it's lacking a few features and we have not yet built workarounds (also, we have not yet set up cross-browser testing which is a pre-requisite to proper [...]

Thank you @amacneil for your prompt feedback. And sorry for the late reply.

As for today, the online version at studio.foxglove.dev does work fine, indeed! It does point out that the only supported browser is Chrome, but seems functional (not used much at this point).

ic, 2022-02-09:
> Per instructions in the readme, I just tried running `docker run -it --rm -p "8080:8080" ghcr.io/foxglove/studio:latest` and visiting localhost:8080 in Firefox 96.0.1, and it works for me. What data source are you loading into Studio?

I could not reach the data loading step :-D

If working in 96.x, then a problem with the version I am using (beta 97).

Given your feedback, please discard my comment, as it should be transient issue.

We are a handful on Firefox interested in Studio, and will watch any call for testers.

amacneil, 2022-02-09:
I think those errors are either happening because you're using a beta version of Firefox, or because of some extension you have installed. I would be curious to hear if it works in the latest stable version of Firefox with no extensions loaded?

TBK, 2022-04-20:
I can confirm that it works perfectly fine with Microsoft Edge (100.0.1185.44) on Windows 11.

Newer Edge versions are based on Chromium.

amacneil, 2022-04-20:
Yeah we should probably list Edge as officially supported and remove the warnings, since it's all Chromium under the hood.

PR #1502 (2021-07-20): Add banner for unsupported browsers + outdated Chrome versions
**User-Facing Changes**
Alert users to when their Chrome version is outdated, or when they're using an unsupported browser (i.e. not Chrome).

**Description**
Example of banner in Safari:

[image]

Resolves #1511.
PR #1514 (2021-07-21): Display only the version banner for browser versions that crash on App render
**User-Facing Changes**
User should be able to see partly working app on browsers outside of Chrome (supported) and Safari 14 (not functional). 

Firefox (not supported, but partly working app) vs. Safari 14 (app not functional):
[image]

**Description**
Does a more generic check for any browser that does not recognize `BigInt64Array` / `BigUint64Array` to render just the banner (i.e. don't load the app at all, as whole page including banner will crash).

Resolve #1511.