foxglove/studio #1538: Need audio support for a cappella robot

samjabrahams, 2021-07-22:
hello

i am building (really planning on building I just need to make a few more design tweaks first) **very sophisticated** a cappella (VOICES ONLY) robots. i have attached prototype designs for these robots below (PLEASE DO NOT TAKE THIS IDEA)

[image]

anyways the robot has audio and right now i dont see a way to play audio from foxglove. it would be nice to hear my acabots sing in foxglove. i didnt see an audio ticket so i made this one.

thx

(Saw your Show HN post! Hope you all are doing well :D)

jtbandes, 2021-07-22:
Good feature idea and great robot idea—with this feature Foxglove Studio could really sing!!

What datatypes do these hypothetical acabots use?

Possibly relevant ROS docs: http://wiki.ros.org/audio_common

samjabrahams, 2021-07-22:
so im personally a fan of raw wav files because i LOVE the clarity that pure uncompressed audio that can provide. my ears are pretty good. but mp3s are nice too some people say you can't tell the difference.

audio common is ok but the main annoying thing is that the raw data and the actual metadata about the audio are in two separate message types!! that means you need two topics to just stream the data unless you know the codec ahead of time. but it is a bit more streamlined for streaming data in chunks since you don't need the overhead of resending the codec on every message. probably not a bad place to start since it seems to be the main "audio" library of ros. you can also always roll your own message types, but now we're getting COMPLICATED 🤷 🤷

samjabrahams, 2021-07-22:
so im personally a fan of raw wav files because i LOVE the clarity that pure uncompressed audio that can provide. my ears are pretty good. but mp3s are nice too some people say you can't tell the difference.

audio common is ok but the main annoying thing is that the raw data and the actual metadata about the audio are in two separate message types!! that means you need two topics to stream the data unless you know the codec ahead of time. but it is a bit more streamlined for streaming data in chunks since you don't need the overhead of resending the codec on every message. probably not a bad place to start since it seems to be the main "audio" library of ros. you can also always roll your own message types, but now we're getting COMPLICATED 🤷 🤷

TobiMiller, 2022-02-16:
any news regarding audio ? what about sending audio to the robot? :)

jtbandes, 2022-02-16:
Hi @TobiMiller, thanks for your interest! No news to share on this feature currently. Can you give more details about specifically what features and data formats you would need?

TobiMiller, 2022-02-17:
@jtbandes Thank you for your answer. For our research project, we need a two-way audio communication to the robot. Here we want a teleoperator to be able to talk to humans around the robot and have them respond as well. So far we have audio_capture and audio_play nodes running on each side and could also communicate via a first attempt to write our own web interface. But meanwhile I think Floxglove would be a much nicer solution for this. If a plugin is available for "audio_common" messages. As I said. Two-way audio communication is what we need.

TobiMiller, 2022-02-18:
@jtbandes does the foxglove dev team have time for such a plugin? Otherwise we would try to build a prototype ourselves. However, none of us in the team masters typescript. so it would take some time. we would appreciate it very much if it is no effort for you to wait for now.

amacneil, 2022-02-18:
This isn't currently on our roadmap - we usually prioritize features based on number of people asking for them.

If you create your own plugin though, we're happy to offer help/advice! ([join our Slack](https://foxglove.dev/slack) for live support)