foxglove/studio #1551: Webfonts

2metres, 2021-07-23:
Given there seems to be an issue using system fonts on Windows. I figured it was best that we select a Web fonts to use for both our sans-serif and monospace fonts.

2metres, 2021-07-23:
### Inter & Roboto Mono
[image]

2metres, 2021-07-23:
### PT Sans/Mono
[image]

2metres, 2021-07-23:
Roboto Sans/Mono
[image]

2metres, 2021-07-23:
### Source Sans/Code Pro
[image]

2metres, 2021-07-23:
### Space Grotesk/Mono
[image]

2metres, 2021-07-23:
Plex Sans/Mono
[image]

amacneil, 2021-07-23:
Source Sans is nice. Good spacing and easy to read.

I prefer Roboto Mono over Code Pro. Code Pro looks to vertically squashed.

Something is wrong with your screenshots because the monospace in `Roboto Sans/Mono` is different from the monospace in `Inter & Roboto Mono`.

jtbandes, 2021-07-23:
The Source Sans screenshot looks like system-ui to me...

2metres, 2021-07-24:
I'm a fan of IBM's Plex Sans/Mono

amacneil, 2021-07-26:
Of these, my least favorite is Space Grotesk. It's trying to be too much.