foxglove/studio #1552: Windows version initializes with blank screen

mikepianka, 2021-07-23:
**Description**
I just tried installing and running the Windows version. When I open the app, I see some content briefly flash but then the window goes blank.

- Version: latest
- OS: Windows 10 19043.1083

**Steps To Reproduce**
Install the Windows version and run it. The window opens, flashes some content, then goes blank like this:

[image]

**Expected Behavior**

**Actual Behavior**

defunctzombie, 2021-07-23:
@mikepianka If you open developer tools (View -> Advanced -> Toggle Developer Tools) are there any obvious errors?

mikepianka, 2021-07-23:
After restarting my machine I reopened Studio and it displayed as expected. But, after closing Studio and reopening it, I get the blank screen again. Developer tools does not bring up anything because it says it has been disconnected.

[image]

defunctzombie, 2021-07-23:
@mikepianka Thanks for doing that troubleshooting with me - when devtools shows disconnected that means an internal process crashed (v8 or the web process to be more specific). We have an theory on what this might be - seems to be related to a font rendering bug we're seeing in our application crash reports after a recent font change. We'll get a fix today and get you a version to test to see if it resolves your issue.

amacneil, 2021-07-23:
Are you using a HiDPI/retina screen on windows by any chance? In the past when we’ve seen this that was a contributing factor.

mikepianka, 2021-07-23:
Ok thanks for the quick response. I'm just using some standard HD monitors, nothing fancy.

amacneil, 2021-07-23:
@mikepianka can you try v0.13.3 which we just released?

amacneil, 2021-07-26:
Fixed in https://github.com/foxglove/studio/commit/c5f97f2619cec40118132f7ab48f2b36c3493c91

closed 2021-07-26 by amacneil

mikepianka, 2021-07-26:
Thanks for the fix. I have been away from my workstation and haven't had a chance to try it yet. I will give it a go tomorrow.

mikepianka, 2021-07-27:
It's fixed and working as expected now. Thanks again 👏

amacneil, 2021-07-27:
Excellent!