foxglove/studio #1566: Middleware-agnostic websocket protocol

amacneil, 2021-07-27:
Studio currently supports [rosbridge websocket](http://wiki.ros.org/rosbridge_suite) connections, which is great for teams using ROS. However, we would like to equally support teams using custom (non-ROS) middlewares.

It is currently possible for non-ROS teams to emulate a [rosbridge server](http://wiki.ros.org/rosbridge_server), but this requires serializing their data as [ROS messages](http://wiki.ros.org/msg) which introduces additional cognitive overhead.

To better support non-ROS teams, we could add support for a neutral Foxglove-defined connection protocol.

- The protocol could support both live and historical data sources (either or both at once if a server has Tivo/DVR features). For simplicity, we could start with only live support.
- For transport, the protocol could use websockets. This is widely compatible (works in browser), easy to secure (TLS & http auth), and reliable (vs UDP etc, which might be more useful in some less common environments). Alternative approaches to consider include HTTP/2 or gRPC (the latter would limit our serialization options).
- For serialization, the client could potentially support multiple formats (e.g. [CBOR](https://cbor.io/), [MessagePack](https://msgpack.org/), [Avro](https://avro.apache.org/), [Protobuf](https://developers.google.com/protocol-buffers), [ROS msg](http://wiki.ros.org/msg)). For simplicity, we could start with only CBOR support.

We would build the protocol client into Studio, offer a reference implementation server (likely in TypeScript), and support community developed server implementations in each language (c++, go, rust, etc), or over time develop our own language-specific implementations.

Out of scope for this feature request is defining new semantic message structures. Each Studio panel requires messages in a particular structure (today these are largely based on existing ROS standards), and it would be up to the server to conform to these expectations.

KerstinKeller, 2021-08-17:
Just stumbled upon Foxglove studio and it seems very promising. I would be interested to connecting it to our middlware solution [eCAL](https://github.com/continental/ecal), which mainly uses Protobuf as serialization. However it would be (easily) possible to just send JSON, too.
Deserializing protobuf can be tricky, especially in the case of non standardized datatypes.

Is this issue just a thought or are you already actively working on an implementation?

amacneil, 2021-08-17:
We're not actively working on this ticket, but happy to collaborate if you are interested.

If you are using the desktop version of Foxglove Studio you could skip the websocket layer and connect directly to eCAL over TCP/UDP (as we do for some existing data sources). However, websockets are nice because they work in the browser, and are often easier to traverse network firewalls etc.

I would avoid sending JSON, it's pretty inefficient for large amounts of data. Sending CBOR or protobuf over the websocket would work. If we support protobuf we would need a way for the server (robot etc) to send the client (Studio) its message definitions at runtime so that we can deserialize them correctly.

The last issue is that once we have deserialized a message, we need to know how to interpret it. Some panels (such as Raw Message panel, Plot panel) work with any message structure. Other panels (Image panel, 3d panel) require specific message structures (today we mostly require ROS formats, for example a [Marker message](http://docs.ros.org/en/noetic/api/visualization_msgs/html/msg/Marker.html) in the 3d panel). Even if the message is serialized as CBOR or protobuf, we need it to have those fields so that we can interpret and visualize it.

If you want to chat more about it in real time feel free to [join our Slack](https://foxglove.dev/slack).

jtbandes, 2021-12-01:
Outstanding issues:

- https://github.com/foxglove/studio/issues/2236

amacneil, 2021-12-13:
This feature is live in [v0.24.0](https://github.com/foxglove/studio/releases/tag/v0.24.0). We can track follow up work in separate tickets.

closed 2021-12-13 by amacneil