foxglove/studio #1574: Resizing / adding panels while Image panels are receiving data sometimes results in blank screen

esthersweon, 2021-07-28:
**Description**
Foxglove sometimes goes to black screen when resizing / adding panels. Seems to occur often when resizing image panels receiving a stream of images

- Version: 0.13.2
- OS: macOS

**Steps To Reproduce**

This issue might be specific to a particular setup (X-forwarding over SSH).

Steps to repro:
1. Install Foxglove Studio on a remote docker container.
2. SSH into docker container with X-forwarding.
3. Launch Foxglove inside docker container. A window should appear on the local machine in which the issue is visible.

**Expected Behavior**
Image panel should continue displaying images, even while user is resizing the panel or adding other panels to the layout.

**Actual Behavior**
Foxglove sometimes goes to black screen when resizing / adding panels. Seems to occur often when resizing image panels receiving a stream of images.

amacneil, 2021-07-29:
Did you ssh into the docker container from macOS host? Last time I attempted to use Studio inside a docker container it failed to even start (https://github.com/foxglove/studio/issues/343).

esthersweon, 2021-08-03:
@amacneil - Updated with appropriate OS / versions. Reported by Peter and Joshua (UC Berkeley).

amacneil, 2021-11-19:
We have not been able to reproduce this lately. If folks encounter this again, please create a new bug with minimal steps to reproduce.

It would be particularly important to know whether this can be reproduced outside of a docker container or not.

closed 2021-11-19 by amacneil