foxglove/studio #1589: First entry in Add Panel is highlighted

rgov, 2021-08-03:
**Description**

When I open the Add Panel sidebar, the first entry in the list, "3D", is highlighted by default. This makes it read like it is a grouping header. It stays highlighted even if you mouseover other entries.

[image]

- Version: 0.14.0
- OS: macOS

jtbandes, 2021-08-03:
FYI, it's highlighted because the list supports keyboard navigation. You can press up/down arrow keys to select items. Not saying this is ideal, just an explanation of why it's like this.

rgov, 2021-08-03:
On Mac apps, such as the Finder, the first time you hit <kbd>↓</kbd> it highlights the first row. If you hit <kbd>↑</kbd> it highlights the last row.

amacneil, 2021-08-03:
> On Mac apps, such as the Finder, the first time you hit ↓ it highlights the first row. If you hit ↑ with no prior selection it highlights the last row.

Agreed, this would be better behavior than highlighting it by default

defunctzombie, 2021-08-05:
Should this also un-highlight the first item when you've used the mouse to hover over another item? Then keyboard keys would navigate from the hovered item?

amacneil, 2022-02-02:
Still a problem

PR #2771 (2022-02-03): Avoid automatically highlighting the first entry in the Add Panel List
**User-Facing Changes**

When showing the add panel list, no entries are highlighted by default. If a user types in the search box, the first entry is highlighted - allowing for quickly filtering and pressing "enter" to add the panel.

When the user clears the search box, highlighting is removed.

Keyboard navigation works as before - with "key down" moving down the list and "key up" moving up. Both wrapping around at each end of the list.

Fixes: #1589
commit 58372d41 (2022-02-03): Avoid automatically highlighting the first entry in the Add Panel List (#2771)

When showing the add panel list, no entries are highlighted by default. If a user types in the search box, the first entry is highlighted - allowing for quickly filtering and pressing "enter" to add the panel.

When the user clears the search box, highlighting is removed.

Keyboard navigation works as before - with "key down" moving down the list and "key up" moving up. Both wrapping around at each end of the list.

Fixes: #1589
closed 2022-02-03 by defunctzombie