foxglove/studio #1597: Getting a TypeError: Do not know how to serialize a BigInt

cjds, 2021-08-05:
**Getting a TypeError: Do not know how to serialize a BigInt**

- Version: 0.14.0
- OS: Ubuntu 20.04

**Steps To Reproduce**

This is the bag

https://drive.google.com/file/d/1NS_cNT8hYqapDandG8VIXOKWq67xBpeF/view?usp=sharing

Viewing the topic `/executor/execute/goal` in the raw messages channel

[image]

Traceback

```
Error stack:
JSON.stringify (<anonymous>)
Ft (file:///opt/Foxglove%20Studio/resources/app.asar/renderer/8074.js:32:5480)
children (file:///opt/Foxglove%20Studio/resources/app.asar/renderer/8074.js:32:8277)
n.value (file:///opt/Foxglove%20Studio/resources/app.asar/renderer/8074.js:23:8513)
n.value (file:///opt/Foxglove%20Studio/resources/app.asar/renderer/8074.js:23:9071)
Dd (file:///opt/Foxglove%20Studio/resources/app.asar/renderer/main.js:429:69495)
L1 (file:///opt/Foxglove%20Studio/resources/app.asar/renderer/main.js:429:69267)
Gd (file:///opt/Foxglove%20Studio/resources/app.asar/renderer/main.js:433:10490)
K1 (file:///opt/Foxglove%20Studio/resources/app.asar/renderer/main.js:433:1011)
qh (file:///opt/Foxglove%20Studio/resources/app.asar/renderer/main.js:433:938)
vf (file:///opt/Foxglove%20Studio/resources/app.asar/renderer/main.js:433:788)
Wd (file:///opt/Foxglove%20Studio/resources/app.asar/renderer/main.js:431:11600)
file:///opt/Foxglove%20Studio/resources/app.asar/renderer/main.js:429:40480
y.unstable_runWithPriority (file:///opt/Foxglove%20Studio/resources/app.asar/renderer/main.js:516:3813)
Qa (file:///opt/Foxglove%20Studio/resources/app.asar/renderer/main.js:429:40248)
eu (file:///opt/Foxglove%20Studio/resources/app.asar/renderer/main.js:429:40422)
$c (file:///opt/Foxglove%20Studio/resources/app.asar/renderer/main.js:429:40356)
y0 (file:///opt/Foxglove%20Studio/resources/app.asar/renderer/main.js:431:8999)
Object.enqueueSetState (file:///opt/Foxglove%20Studio/resources/app.asar/renderer/main.js:429:44642)
n.h.setState (file:///opt/Foxglove%20Studio/resources/app.asar/renderer/main.js:483:1329)
file:///opt/Foxglove%20Studio/resources/app.asar/renderer/8074.js:23:7726
Fe (file:///opt/Foxglove%20Studio/resources/app.asar/renderer/main.js:3745:12112)
Component stack:
n (file:///opt/Foxglove%20Studio/resources/app.asar/renderer/8074.js:23:6569)
span
li
i (file:///opt/Foxglove%20Studio/resources/app.asar/renderer/9856.js:7:654)
r (file:///opt/Foxglove%20Studio/resources/app.asar/renderer/9856.js:6:5579)
ul
li
p (file:///opt/Foxglove%20Studio/resources/app.asar/renderer/9856.js:6:1672)
s (file:///opt/Foxglove%20Studio/resources/app.asar/renderer/9856.js:6:8387)
r (file:///opt/Foxglove%20Studio/resources/app.asar/renderer/9856.js:6:5579)
ul
li
p (file:///opt/Foxglove%20Studio/resources/app.asar/renderer/9856.js:6:1672)
s (file:///opt/Foxglove%20Studio/resources/app.asar/renderer/9856.js:4:8315)
r (file:///opt/Foxglove%20Studio/resources/app.asar/renderer/9856.js:6:5579)
ul
li
p (file:///opt/Foxglove%20Studio/resources/app.asar/renderer/9856.js:6:1672)
s (file:///opt/Foxglove%20Studio/resources/app.asar/renderer/9856.js:6:8387)
r (file:///opt/Foxglove%20Studio/resources/app.asar/renderer/9856.js:6:5579)
ul
li
p (file:///opt/Foxglove%20Studio/resources/app.asar/renderer/9856.js:6:1672)
s (file:///opt/Foxglove%20Studio/resources/app.asar/renderer/9856.js:6:8387)
r (file:///opt/Foxglove%20Studio/resources/app.asar/renderer/9856.js:6:5579)
ul
li
p (file:///opt/Foxglove%20Studio/resources/app.asar/renderer/9856.js:6:1672)
s (file:///opt/Foxglove%20Studio/resources/app.asar/renderer/9856.js:6:8387)
r (file:///opt/Foxglove%20Studio/resources/app.asar/renderer/9856.js:6:5579)
ul
R (file:///opt/Foxglove%20Studio/resources/app.asar/renderer/9856.js:10:1820)
div
file:///opt/Foxglove%20Studio/resources/app.asar/renderer/main.js:3769:1499
div
file:///opt/Foxglove%20Studio/resources/app.asar/renderer/main.js:3769:1499
ct (file:///opt/Foxglove%20Studio/resources/app.asar/renderer/8074.js:32:6453)
w (file:///opt/Foxglove%20Studio/resources/app.asar/renderer/main.js:3769:861)
div
file:///opt/Foxglove%20Studio/resources/app.asar/renderer/main.js:3769:1499
p (file:///opt/Foxglove%20Studio/resources/app.asar/renderer/main.js:4032:5707)
de (file:///opt/Foxglove%20Studio/resources/app.asar/renderer/main.js:4032:9286)
Suspense
div
div
J (file:///opt/Foxglove%20Studio/resources/app.asar/renderer/main.js:441:14368)
N (file:///opt/Foxglove%20Studio/resources/app.asar/renderer/main.js:412:2550)
N (file:///opt/Foxglove%20Studio/resources/app.asar/renderer/main.js:412:2550)
J (file:///opt/Foxglove%20Studio/resources/app.asar/renderer/main.js:441:19765)
div
div
g (file:///opt/Foxglove%20Studio/resources/app.asar/renderer/main.js:441:11861)
div
P (file:///opt/Foxglove%20Studio/resources/app.asar/renderer/main.js:441:7161)
w (file:///opt/Foxglove%20Studio/resources/app.asar/renderer/main.js:3769:861)
H (file:///opt/Foxglove%20Studio/resources/app.asar/renderer/main.js:4040:282)
k (file:///opt/Foxglove%20Studio/resources/app.asar/renderer/main.js:4040:1606)
div
_ (file:///opt/Foxglove%20Studio/resources/app.asar/renderer/main.js:4:21340)
file:///opt/Foxglove%20Studio/resources/app.asar/renderer/main.js:4071:10116
p (file:///opt/Foxglove%20Studio/resources/app.asar/renderer/main.js:4071:10176)
w (file:///opt/Foxglove%20Studio/resources/app.asar/renderer/main.js:3769:861)
div
div
g (file:///opt/Foxglove%20Studio/resources/app.asar/renderer/main.js:441:11861)
div
P (file:///opt/Foxglove%20Studio/resources/app.asar/renderer/main.js:441:7161)
div
di (file:///opt/Foxglove%20Studio/resources/app.asar/renderer/main.js:524:7993)
div
_ (file:///opt/Foxglove%20Studio/resources/app.asar/renderer/main.js:4:21340)
fo (file:///opt/Foxglove%20Studio/resources/app.asar/renderer/main.js:4188:79)
div
p (file:///opt/Foxglove%20Studio/resources/app.asar/renderer/main.js:4032:5707)
mv (file:///opt/Foxglove%20Studio/resources/app.asar/renderer/main.js:4188:5201)
x2 (file:///opt/Foxglove%20Studio/resources/app.asar/renderer/main.js:4195:64854)
Suspense
file:///opt/Foxglove%20Studio/resources/app.asar/renderer/main.js:411:1311
t (file:///opt/Foxglove%20Studio/resources/app.asar/renderer/main.js:4239:10606)
be (file:///opt/Foxglove%20Studio/resources/app.asar/renderer/main.js:4032:2817)
qr (file:///opt/Foxglove%20Studio/resources/app.asar/renderer/main.js:4195:34730)
v2 (file:///opt/Foxglove%20Studio/resources/app.asar/renderer/main.js:4195:60966)
f2 (file:///opt/Foxglove%20Studio/resources/app.asar/renderer/main.js:4195:60333)
E_ (file:///opt/Foxglove%20Studio/resources/app.asar/renderer/main.js:4195:56621)
o2 (file:///opt/Foxglove%20Studio/resources/app.asar/renderer/main.js:4195:58481)
s2 (file:///opt/Foxglove%20Studio/resources/app.asar/renderer/main.js:4195:58676)
l (file:///opt/Foxglove%20Studio/resources/app.asar/renderer/main.js:4078:9626)
t (file:///opt/Foxglove%20Studio/resources/app.asar/renderer/main.js:4239:10606)
m (file:///opt/Foxglove%20Studio/resources/app.asar/renderer/main.js:4078:8331)
l (file:///opt/Foxglove%20Studio/resources/app.asar/renderer/main.js:4078:5703)
Ou (file:///opt/Foxglove%20Studio/resources/app.asar/renderer/main.js:4195:39314)
J_ (file:///opt/Foxglove%20Studio/resources/app.asar/renderer/main.js:4195:50850)
l2 (file:///opt/Foxglove%20Studio/resources/app.asar/renderer/main.js:4195:59886)
ef (file:///opt/Foxglove%20Studio/resources/app.asar/renderer/main.js:4195:38976)
p (file:///opt/Foxglove%20Studio/resources/app.asar/renderer/main.js:4032:5707)
M2 (file:///opt/Foxglove%20Studio/resources/app.asar/renderer/main.js:4195:67777)
Bu (file:///opt/Foxglove%20Studio/resources/app.asar/renderer/main.js:3751:10133)
Wx (file:///opt/Foxglove%20Studio/resources/app.asar/renderer/main.js:4195:105894)
Au (file:///opt/Foxglove%20Studio/resources/app.asar/renderer/main.js:3751:5982)
Ua (file:///opt/Foxglove%20Studio/resources/app.asar/renderer/main.js:3751:13219)
mc (file:///opt/Foxglove%20Studio/resources/app.asar/renderer/main.js:3751:8799)
eu (file:///opt/Foxglove%20Studio/resources/app.asar/renderer/main.js:3751:7490)
z (file:///opt/Foxglove%20Studio/resources/app.asar/renderer/main.js:469:23504)
Hx (file:///opt/Foxglove%20Studio/resources/app.asar/renderer/main.js:4195:106728)
p (file:///opt/Foxglove%20Studio/resources/app.asar/renderer/main.js:4032:5707)
w (file:///opt/Foxglove%20Studio/resources/app.asar/renderer/main.js:3769:861)
y.f6 (file:///opt/Foxglove%20Studio/resources/app.asar/renderer/main.js:524:11846)
file:///opt/Foxglove%20Studio/resources/app.asar/renderer/main.js:4195:104485
file:///opt/Foxglove%20Studio/resources/app.asar/renderer/main.js:4195:71071
Nx (file:///opt/Foxglove%20Studio/resources/app.asar/renderer/main.js:4195:104558)
wu (file:///opt/Foxglove%20Studio/resources/app.asar/renderer/main.js:3751:15436)
```

**Expected Behavior**

**Actual Behavior**

defunctzombie, 2021-08-05:
@cjds 

I opened the referenced bag and setup a Raw Message panel with `/executor/execute/goal` as shown in the screenshot below.

[image]

As I play the bag I was able to see the message. I did not encounter an error.

Were you using that same path in the raw message panel or maybe something else?

closed 2021-08-05 by jtbandes

cjds, 2021-08-05:
I was expanding every subfield in that bag when I expanded to

goal->navigation->config_params[0]

Is when I got the error

jtbandes, 2021-08-05:
This is now fixed in `main` if you are running from source. It will be available in our next release, later this week or early next week – keep an eye on https://github.com/foxglove/studio/releases/

cjds, 2021-08-05:
Awesome. Thanks :-D

PR #1599 (2021-08-05): Handle bigints in getValueActionForValue generated filter strings
**User-Facing Changes**
Fixed a crash when hovering on int64/uint64 values inside arrays in Raw Messages. (#1597)

**Description**
Fixes #1597 by checking for `typeof value === "bigint"`.
(This was partly fixed in https://github.com/foxglove/studio/pull/1297, but a unit test wasn't added to cover the bigint case — if it had this issue would have come up as well.)

Related: #1445.
commit ee0ac552 (2021-08-05): Handle bigints in getValueActionForValue generated filter strings (#1599)

**User-Facing Changes**
Fixed a crash when hovering on int64/uint64 values inside arrays in Raw Messages. (#1597)

**Description**
Fixes #1597 by checking for `typeof value === "bigint"`.
(This was partly fixed in https://github.com/foxglove/studio/pull/1297, but a unit test wasn't added to cover the bigint case — if it had this issue would have come up as well.)

Related: #1445.