foxglove/studio #1612: Unhandled exception: ResizeObserver loop limit exceeded

amacneil, 2021-08-09:
**Description**
We are getting thousands of these errors reported in Sentry.

As far as I understand this error doesn't have any user facing impact. If it's possible to prevent the error from happening, then we should do so. If not, we should skip reporting them via Sentry's [ignoreErrors](https://docs.sentry.io/clients/javascript/config/) setting.

If we are going to ignore the error, we should do it client side rather than in Sentry UI, to avoid sending unnecessary data and also so that anyone configuring sentry to point at a different project will share the config.

jtbandes, 2021-08-09:
See also: https://github.com/WICG/resize-observer/issues/38#issuecomment-422126006

defunctzombie, 2021-08-19:
https://github.com/foxglove/studio/pull/1645

jtbandes, 2021-08-19:
[image]

amacneil, 2021-08-23:
Still happening

defunctzombie, 2021-08-24:
I've suppressed it further for now but keeping this issue open until we figure out which parts of our resize observers cause this error. We should be able to configure them in a way to avoid this error. If we learn otherwise then we can close this out and leave the suppressions.