foxglove/studio #1615: Enable choosing the NavSatFix topic

akin-sisbot-tm, 2021-08-09:
Currently Foxglove assumes that there will be only one topic of type sensor_msgs/NavSatFix. However in some cases involving multiple GPS units or multiple vehicles there might be multiple topics of the same type. 

It would be great if the UI would allow either choosing the topic for the Map panel or even better, the UI displays multiple NavSatFix topics in the same Map panel.

valschmidt, 2021-10-07:
I think this is a great feature suggestion, and would prefer display of all NavSatFix topics

defunctzombie, 2021-10-07:
I've started work on initial support here https://github.com/foxglove/studio/pull/1926 - Will try to get the basic feature landed for the next release and we can iterate from there.

commit 56ba3630 (2021-10-08): Feature: add control to map panel to toggle topics (#1926)

Add topics to the layer control so a user can toggle individual topics on/off in the map display. The disabled topics are persisted.

Fixes: #1615