foxglove/studio #1622: Amplitude metrics not working on web build

esthersweon, 2021-08-11:
**Description**
Amplitude metrics are not working on studio.foxglove.dev.

- Version: 0.14.0
- OS: macOS

**Steps To Reproduce**

- Navigate to studio.foxglove.dev
- Look at Chrome console for errors
- See following Amplitude errors
[image]

**Expected Behavior**
No Amplitude errors in console; metrics correctly collected from web build.

**Actual Behavior**
Errors in console; metrics not being collected from web build.

jtbandes, 2021-08-11:
This bug is caused by https://github.com/webpack/webpack/issues/12960.

amacneil, 2021-09-01:
Pretty sure this is still not working

jtbandes, 2021-09-02:
One issue now is that in the web build, the app version is reported as 0.0.0:

https://github.com/foxglove/studio/blob/1d34d9672fbc8e4968f72c53c708cf9c03693c52/packages/studio-base/src/services/AmplitudeAnalytics.ts#L79-L81

(I don't think this is what's causing analytics to not show up at all though)

amacneil, 2021-09-08:
@esthersweon did this get fixed? If not, would be great if someone can look into this this week.

jtbandes, 2021-09-17:
Indeed, I don't think any work was done on this since my last comment (at least I didn't see any PRs go by)

PR #1638 (2021-08-13): Fix Amplitude analytics in web build
**User-Facing Changes**
None

**Description**
Fixes #1622 by working around the webpack bug that causes it.
commit 97e55e77 (2021-08-13): Fix Amplitude analytics in web build (#1638)

**User-Facing Changes**
None

**Description**
Fixes #1622 by working around the webpack bug that causes it.
closed 2021-09-18 by amacneil