foxglove/studio #1624: Is there a way to send rosrun or roslaunch commands/kill them from the foxglove studio?

sisaha9, 2021-08-12:
I don't see a feature like this existent but if it can be made that would be great and really enable using Foxglove Studio as a full control interface. Currently it seems publishing messages to rostopics and using variables is the best way to enable it from our end

amacneil, 2021-08-13:
It's not currently possible. It's an interesting feature request though.

How would you expect this to work? Would you want a full terminal emulator in studio (similar to vscode) or more of just a start/stop button? If a start/stop button, how would you want to select the launch file, and where would you want the stdout to go?

Are you running ROS on the same computer that your robot is running on (localhost), or are you connected over a network? If remote, do you have ssh access to it? How would you want studio to connect and authenticate to the remote robot?

sisaha9, 2021-08-18:
This feature is probably what supports my use case the best: https://github.com/foxglove/studio/issues/1662

I shouldn't have used rosrun or roslaunch since I was intending this for ROS2. Ideally we should be able to make use of SSH access to connect to the machine remotely and use. I imagine by using different parameters and having nodes listen to that topic we can best set up a control station

amacneil, 2021-08-23:
I don't see how #1662 is related to your request in this issue (running roslaunch from within studio). Setting parameters is nice, but ros has to be already running before you can use that.

defunctzombie, 2022-07-07:
I'm gonna close this as not planned. Its an interesting idea but not something on our roadmap.

closed 2022-07-07 by defunctzombie