foxglove/studio #1629: Display nav_msgs/Odometry messages in the 3D panel

esthersweon, 2021-08-12:
Requested by Studio user in Slack; visualizing this is currently possible in rviz.

Example: 
[image]

geetesh, 2021-08-13:
yes, please add support for nav_msgs

jtbandes, 2021-08-13:
@geetesh which nav_msgs are you most in need of?

geetesh, 2021-08-13:
@jtbandes `nav_msgs/Odometry`
thank you

PhilippSchmaelzle, 2022-08-04:
Hi, I just started using foxglove and it works great! Appreciate your efforts.

Visualization of `nav_msgs/Odometry` would be great. Including the Covariance ellipsoids.
I use it for the robot_localization package with multiple global positioning systems. There it helps to see which device provides which information.

Best 
Philipp

ashwinvkNV, 2022-12-02:
Same, it would be great to add support for `nav_msgs/Odometry`. Especially if a user is expected to use foxglove for visualizing a state estimation pipeline.

belorenz, 2022-12-13:
Visualization of nav_msgs/Odometry would be highly appreciated!

jtbandes, 2022-12-13:
The other issue was just closed as a duplicate of this issue. However, be advised that we will likely migrate all feature requests to Discussions soon, so that we can keep Issues for bug reports and actively planned work.

jtbandes, 2022-12-13:
Also, the recently announced [message converter extensions](https://foxglove.dev/blog/announcing-message-converter-extensions-in-foxglove-studio) can be useful if you are trying to customize the visualization of data like Odometry. Check it out!