foxglove/studio #1641: (Panel: Plot) Not Working Properly

sink0918, 2021-08-14:
**Description**
As I add a data into the plot, the plot starts drawing slower and slower, eventually can't work at all.

- Version:  v0.15.0
- OS: Studio in Windows 11 with ROS noetic in WSL 2 

**Steps To Reproduce**

Connect to ROS 1 bridge then add anything to the plot

**Expected Behavior**
When I using plot in the webviz

https://user-images.githubusercontent.com/57067221/129442644-4bc0ee48-ba4e-4446-902d-9c51b86e4eff.mp4

**Actual Behavior**
When I using plot in the Studio, it is already crushed

https://user-images.githubusercontent.com/57067221/129442683-0e6a3b8a-bab0-431a-a0a1-a30e327e2fcb.mp4

defunctzombie, 2021-08-14:
@sink0918 Thanks for reporting this! We'll take a look.

sink0918, 2021-08-24:
The problem still exists on v0.15.0 when I turn "X range in seconds" to 5 in the plot panel.

https://user-images.githubusercontent.com/57067221/130574423-095125bb-9a18-41e8-9c14-eeb8f735b1ac.mp4

defunctzombie, 2021-08-24:
@sink0918 Whats the frequency of `/vision/finally_data.energy_sin_vel` ? How big is the `/vision/finally_data` message itself (i.e. what are the other fields).

sink0918, 2021-08-24:
> @sink0918 Whats the frequency of `/vision/finally_data.energy_sin_vel` ? How big is the `/vision/finally_data` message itself (i.e. what are the other fields).

In the Studio setting, I turn message_rate to 60 Hz 
[image]

And this is the info of `/vision/finally_data` 
[image]

BTW, when I use webviz, all the data added to the plot can be fluently drawn in the exact same settings.

defunctzombie, 2021-08-24:
@sink0918 Do you mind sharing the message definition for `vision/Data`? I'd like replicate sending a similar message type at 100hz locally. I did similar tests with my own messages (thats how I landed on the referenced fix) but maybe there is more to it based on the message.

If you prefer to share the message definition in private you can do so directly to me via our slack or email.

sink0918, 2021-08-24:
> @sink0918 Do you mind sharing the message definition for `vision/Data`? I'd like replicate sending a similar message type at 100hz locally. I did similar tests with my own messages (thats how I [...]

I have send the email to contact@foxglove.dev, please check.

defunctzombie, 2021-08-28:
@sink0918 I looked into the slowdown issue while connected to a live stream. The slowdown comes from the underlying charting library (chartjs) we are using for the plot.

The library has two areas which contribute to the slowdown as more and more data is plotted in realtime. One of these is the rendering of the points. Each point is individually rendered and, as the number of points increases on the plot, the time to render the points increases. This issue also impacts webviz; it just takes a little longer than in studio due to the next issue area which is a change in plugin notification handling in chartjs. Webviz uses chartjs v2.9. In studio, we've upgraded to chartjs v3. From my profiling of studio, chartjs v3 shows a new slowdown in an internal codepath which delivers notifications to plugins. We use two plugins for our plots (datalabels and zoom). While we could look at removing these or finding alternatives it isn't quick and not necessarily likely to fix the underlying chartjs notification slowdown.

These two issues: point rendering and plugin notifications are what contribute to the slowdown as more and more data changes on the streaming chart. We can alleviate the point rendering issue by making point rendering a plot option. This will delay when you start to see rendering latency but also impacts features like hovering to see point data.

Unfortunately there is no quick win here like there was with disabling chart animations which were a contributor to the slowdown issue that impacted the sliding window view. For now I recommend using a window size that provides the streaming experience you are comfortable with. I will keep this issue open and this comment will provide further performance guidance when we are able to spend time looking into this.

sink0918, 2021-08-28:
Yes, I think the point rendering option may help. Like the `rqt_plot` also have a similar option.

[image]

jessicaaustin, 2021-09-29:
FWIW, I've seen similar performance issues with "lots" of plots in a single layout using the default settings of 60Hz for the refresh rate (The `Message Rate` Studio setting). Lowering the refresh rate fixes the problem so far, and that's fine for me because I don't need 60Hz update rate for my use case.

An example of what I mean by "a lot":

[image]

All the plots here are from topics updating at 20Hz, with a 10s rolling window, and I'm connected directly to the ROS graph. If I use 60Hz then some of the plots will "stick" and fail to update for a few seconds before catching up. At 5Hz everything works fine for 10s-30s windows.

If/when this gets picked up again, happy to test or provide feedback.

amacneil, 2021-11-19:
@defunctzombie suggested closing out this bug since we addressed the main issues, and open a new feature to allow toggling point rendering which improves performance.

Longer term we have discussed moving to webGL plots for better performance.

commit 848b9010 (2021-08-18): Fix: filter plot panel datasets to view bounds

Rather than sending the entire dataset to the PlotChart and subsequent
TimeBasedChart and Chart components, filter the dataset down to the
items in the settings display.

Fixes: #1641
commit a0bcc414 (2021-08-19): Fix: filter plot panel datasets to view bounds

Rather than sending the entire dataset to the PlotChart and subsequent
TimeBasedChart and Chart components, filter the dataset down to the
items in the settings display.

Fixes: #1641
commit 6f315f9c (2021-08-19): Fix: filter plot panel datasets to view bounds

Rather than sending the entire dataset to the PlotChart and subsequent
TimeBasedChart and Chart components, filter the dataset down to the
items in the settings display.

Fixes: #1641
PR #1649 (2021-08-19): Fix: filter plot panel datasets to view bounds
**User-Facing Changes**
When using a streaming data source and a follow window size, the plot panel no longer lags as the datasource continues to stream.

**Description**
Rather than sending the entire dataset to the PlotChart and subsequent
TimeBasedChart and Chart components, filter the dataset down to the
items in the settings display.

Fixes: #1641
closed 2021-11-21 by sink0918