foxglove/studio #1646: Fix bug with changes appearing immediately after switching layouts

jtbandes, 2021-08-16:


jtbandes, 2021-09-02:
Still happens sometimes, especially with newly added panels. Example with splitting an existing Welcome panel from the welcome layout:

https://user-images.githubusercontent.com/14237/131890737-96528128-fe50-4cb3-9466-8d66f835ef22.mov

jtbandes, 2021-09-04:
Got another repro:
- Use "inject edit" feature (on a non-selected layout) which directly updates the server by adding a panel to the layout
- Sync to ensure new layout is downloaded
- Switch to the edited layout – changes appear

https://user-images.githubusercontent.com/14237/132081388-dc4786a1-1ad7-403f-ad00-7c84ba75fc18.mov

defunctzombie, 2021-09-04:
@jtbandes that seems like correct behavior to me. The working version is now different than the base. In this case maybe just the default config, but in other cases maybe more changed. If the server version changes the base on a sync I would expect that to be shown as changes compared to the working copy?

jtbandes, 2021-09-04:
> If the server version changes the base on a sync I would expect that to be shown as changes compared to the working copy?

If you’ve never made changes to the layout locally or even clicked on it, someone else changing it on the server shouldn’t result in a modified dot on your machine. Notice in my video the dot doesn’t appear upon sync, but only when loading the layout.

The dot being visible is just “working copy != undefined”. An unmodified layout doesn’t have a working copy, only the baseline

defunctzombie, 2021-09-10:
From our chat - it seems that the issue here is synthetic and would not occur the same way through user interaction. The "inject edit" adds a panel to the layout but fails to add the default config. A user that is working with the app on two computers and adding a layout would not produce the same result since the app will set the default config when the panel is added to the layout.

I'm gonna close this issue since through user interaction this same flow would not occur. If we do find other instances of changes appearing in unexpected ways (or via remote interaction) we should open new issues with reproduction for those.

commit 82784e4a (2021-08-31): Fix: avoid changing panel state for unchanged config

When a panel saves its config, this triggers a set of reducers
to run. If the config is unchanged from the previous config, the save
config reducer keeps the same state object to indicate there were
no changes to the state.

Fixes #1646
PR #1725 (2021-08-31): Fix: avoid changing panel state for unchanged config

**User-Facing Changes**
When the user changes layouts they would be shown that their layout is "dirty" and has unsaved changes even tho there were no actual changes to the layout. After this change the layout is not marked as dirty when just changing the layout.

**Description**
When a panel saves its config, this triggers a set of reducers
to run. If the config is unchanged from the previous config, the save
config reducer keeps the same state object to indicate there were
no changes to the state.

Fixes #1646
commit 79184578 (2021-08-31): Fix: avoid changing panel state for unchanged config (#1725)

When a panel saves its config, this triggers a set of reducers
to run. If the config is unchanged from the previous config, the save
config reducer keeps the same state object to indicate there were
no changes to the state.

Fixes #1646
PR #1752 (2021-09-02): Fix: save the latest layout state

**User-Facing Changes**
When a user "splits" a panel and then saves the layout their saved panel would still appear as "unsaved" when navigating away/back to the layout. This change fixes that behavior so when a layout is saved after splitting a panel it does not appear unsaved after navigation away/to.

**Description**
Some actions would cause re-entrant behavior to performAction which
resulted in the wrong version of layout state being saved. This change
moves setLayoutState to the end of performAction to serialize updates
to the outgoing layout state.

Fix [... cut]
closed 2021-09-10 by defunctzombie