foxglove/studio #1661: Native ROS 2 Connection: Support publishing back to ROS stack

esthersweon, 2021-08-18:
Studio users connected to their data source via a native ROS 2 connection should be able to publish back to their ROS stack.

jmm-slamcore, 2022-07-27:
Any updates on this issue? Can't seem to be able to publish goals to Nav2 yet/not even able to see the publish pose button on the desktop app when using ROS2.

defunctzombie, 2022-07-28:
We recommend using the rosbridge connection if you need publish support in ROS2. We do not currently have plans to add publishing support to the native ROS2 connection.

jtbandes, 2022-11-30:
Going forward, we plan to deprecate and remove the ROS 2 native connection (https://github.com/foxglove/studio/issues/4900). We are working on a new [foxglove_bridge](https://github.com/foxglove/ros-foxglove-bridge), which will provide improved performance and reliability over the Rosbridge and native connections. You can already use the new bridge today (see the README). Stay tuned for upcoming announcements about the new bridge!

closed 2022-11-30 by jtbandes

pepeRossRobotics, 2022-12-01:
@jtbandes Will foxglove_bridge support UDP connections?

jtbandes, 2022-12-01:
Not in its current form – see https://github.com/foxglove/studio/discussions/4613 for more discussion on UDP.

ruffsl, 2023-03-31:
> Not in its current form – see https://github.com/foxglove/studio/discussions/4613 for more discussion on UDP.

@jtbandes , that link now 404s. Where were these discussions migrated to?

jtbandes, 2023-03-31:
I think it was one of these:
- https://github.com/orgs/foxglove/discussions/15
- https://github.com/orgs/foxglove/discussions/574