foxglove/studio #1662: Native ROS 2 Connection: Support reading and setting ROS params

esthersweon, 2021-08-18:
Studio users connected to a data source via a native ROS 2 connection should be able to:
- Receive and ROS params from their ROS stack
- Set ROS param values by sending them back to the stack

amacneil, 2021-12-15:
@esthersweon can you please split this ticket up into separate tickets for each data source that is missing params support?

Is there anywhere in our docs that tells you which data sources support params?

jhurliman, 2022-01-07:
This should be working in ROS1 native and rosbridge. ROS2 native param support is being tracked in https://github.com/foxglove/studio/issues/2614

closed 2022-01-07 by jhurliman

chenjunnn, 2022-01-11:
> This should be working in ROS1 native and rosbridge.

@jhurliman I'm using rosbridge connection and open the `Parameter` panel, but it still shows `Connect to a ROS source to view parameters`. And I cound not find related PRs or commits. Is this feature not supported in the current version? If so, I would appreciate it if you could tell me that when it would be supported.

jtbandes, 2022-01-11:
Params are supported in ROS 1 native connections since https://github.com/foxglove/studio/pull/601, but not currently in rosbridge connections.

amacneil, 2022-01-13:
@esthersweon maybe somewhere in our docs we could have a matrix of which features are supported by which data sources?