foxglove/studio #1664: can't copy-paste errors from error tooltip

wkalt, 2021-08-18:
I am encountering an error in the 3D panel. Screenshot is attached. It would be useful if I could somehow copy/paste the error from the tooltip but mousing away from the red circle makes the tooltip go away. If I could copy/paste or get in-app instructions to check the dev console logs, it might avoid a needless message exchange with the dev team.

[image]

defunctzombie, 2022-03-17:
The copy-paste of the content works. The problem is we hide the tooltip when the user mouseleave's the target element (the circle icon) and have no notion of the user's entering the tooltip content.

https://github.com/foxglove/studio/blob/main/packages/studio-base/src/components/Tooltip.tsx#L91-L95

defunctzombie, 2022-09-06:
This specific issue is related to old3d error notifications. I just checked using the new settings sidebar and there we are able to properly hover and copy the error text if a settings item is showing an error tooltip.

Closing this out since we are migrating off the 3d panel towards 3d (beta) which uses the new settings apis and does not suffer from this usability issue.

closed 2022-09-06 by defunctzombie