foxglove/studio #1675: Image panel marker dropdown should only show compatible topics

jhurliman, 2021-08-19:
The image panel has a dropdown to select a topic for markers. However, the dropdown is populated with every known topic even though the majority of topics can't be visualized. It should restrict the list to only show topics with datatype `visualization_msgs/ImageMarker` or `foxglove_msgs/ImageMarkerArray`.

amacneil, 2021-08-19:
Is this related to https://github.com/foxglove/studio/pull/1584?

jtbandes, 2021-08-19:
Note that we do already have code to filter by datatype: https://github.com/foxglove/studio/blob/6ab76abd4436939053a6cc4978bf563312b31d7b/packages/studio-base/src/panels/ImageView/index.tsx#L291-L307

And this is used to filter the list of options in the dropdown: https://github.com/foxglove/studio/blob/6ab76abd4436939053a6cc4978bf563312b31d7b/packages/studio-base/src/panels/ImageView/util.ts#L53-L58

However, in addition to the main list, there is **also** a field that lets you "add" your own topic, and I assume that's what John is referring to in this ticket. It looks like that field includes all topics that aren't selected, not just the ones with matching datatypes:

https://github.com/foxglove/studio/blob/6ab76abd4436939053a6cc4978bf563312b31d7b/packages/studio-base/src/panels/ImageView/index.tsx#L460-L465

amacneil, 2022-02-16:
@jtbandes is this still an issue? I have only ever seen marker topics listed in the image marker dropdown.

jtbandes, 2022-02-16:
See the last paragraph of https://github.com/foxglove/studio/issues/1675#issuecomment-902133899

defunctzombie, 2022-02-17:
In addition to restricting the list - maybe it should be an action that opens a dialog to select the topic. Or the dropdown should act more like a multi-select combo box and show the other valid topics that are not selected. The current UI look/feel is strange.

PR #2877 (2022-03-04): Show all available annotation topics in the annotations dropdown
**User-Facing Changes**
Users can select any valid annotation topic using the annotation dropdown. The relevant annotation topics are still first.

[image]

**Description**
This change updates the Image panel annotations dropdown to show all available annotation topics. The "related" topics appear first. Those are the ones that are under the same "camera namespace" (i.e same topic prefix). Then the remaining topics after.
    
This change also switches the Dropdown component to the MUI Select component for the dropdown.
    
Fixe [... cut]
commit 5cfa9039 (2022-03-04): Show all available annotation topics in the annotations dropdown (#2877)

This change updates the Image panel annotations dropdown to show all available annotation topics. The "related" topics appear first. Those are the ones that are under the same "camera namespace" (i.e same topic prefix). Then the remaining topics after.

This change also switches the Dropdown component to the MUI Select component for the dropdown.

Fixes: #1675
closed 2022-03-04 by defunctzombie