foxglove/studio #1683: Ability to link 3d panel markers to source topics

amacneil, 2021-08-23:
Markers in the 3d panel usually represent some underlying message. For example, a marker might represent a tracked object.

We would like to add the ability to link markers with their underlying message, and allow users to inspect the source message. For example, when clicking a marker in the 3d panel, I should be able (via a button or similar) to click through to open the Raw Messages panel showing the relevant topic (e.g. `/tracked_objects{object.id==5980}`).

The link from the marker to the underlying topic can be encoded in the marker. This will require extending the ROS Marker message which we currently use.

jhurliman, 2022-01-08:
This will mostly be achieved by <https://github.com/foxglove/studio/pull/2617>. It's not a link to the Raw Message panel but rather embedding the raw message information in the popup inside the 3D panel.

defunctzombie, 2022-04-08:
@jtbandes assigning to you for inclusion into the BYOD work. We should incorporate this idea into our version of 3d object/marker/annotation.