foxglove/studio #1684: Make ROS message types available to extensions

rgov, 2021-08-23:
https://github.com/foxglove/studio/blob/main/packages/studio-base/src/types/Messages.ts

These types aren't available in the extension API, but at least some like `Header` ought to be so it is easier to define compound message types in extension code.

closed 2022-08-07 by github-actions[bot]

amacneil, 2022-08-07:
Pretty sure this is supported now.

defunctzombie, 2022-08-07:
@amacneil What makes you think it is supported? I don't recall any package we've published with the ROS message types as typescript definitions.

amacneil, 2022-08-08:
Nevermind, I thought this was talking about ROS schemas being available to user scripts.

As for the typescript types, a logical place for them to live would be https://github.com/foxglove/rosmsg-msgs-common