foxglove/studio #1686: Automate publishing releases to apt repo

amacneil, 2021-08-23:
Publishing releases to apt repo is currently a manual step following the release. This should be automated in the [post-release.yml](https://github.com/foxglove/studio/blob/main/.github/workflows/post-release.yml) workflow.

amacneil, 2021-08-27:
Yeah there is a pretty decent github action for normal homebrew, not sure if it supports cask (might need modification). I use it for dbmate:

https://github.com/amacneil/dbmate/blob/main/.github/workflows/release.yml

Doubt there will be any existing PR if we do this in our post-release action.

rgov, 2021-08-28:
> Doubt there will be any existing PR if we do this in our post-release action.

As in, you open a PR for 0.17.0 and then quickly cut a bug-fix 0.17.1 before the 0.17.0 cask update is merged.

amacneil, 2021-08-29:
Oh, I see. Yeah, that would be confusing. I wonder how often that happens (surely pretty frequent given how many apps homebrew/cask support).

closed 2022-01-08 by amacneil