foxglove/studio #1732: Confusing synchronization behavior in plot panels

defunctzombie, 2021-08-31:
**Current Experience**
Timeseries plots (and state transition panels) are pan/zoom synchronized. When a user pan/zooms one timeseries plot, other timeseries plots in the layout also pan/zoom by the same amount. This keeps the timeseries x min/max range the same across all plots. The intent for this feature is to easily allow a user to zoom into one plot to closer inspect one section of timeseries data and have the other plots zoom into the same data for comparison. Without this synchronization the user would need to pan/zoom each plot individually if they want to get a detail view on a specific section of the timeline.

When a user pan/zooms a plot - that plot is considered manually pan/zoomed by the user. The behavior is that when a user goes to pan/zoom another plot panel the first no longer synchronizes. This is shown below:

https://user-images.githubusercontent.com/84792/131562647-c89b275b-9f73-458a-8e25-d8e30ea79d32.mp4

This is a somewhat surprising user experience since just panning the plot doesn't necessarily mean you no longer want panning other plots to no longer pan the first plot.

The "reset view" button does not make it so obvious that clicking it will "re-lock" the panel. The user may want to keep the current view but keep the panel locked for pan/zoom.

**Desired Experience**
The plot panels should remain locked for pan/zoom unless the user takes an explicit action. Maybe a control on the panel itself?

While it may add to the plot view clutter - I like the idea of a control on the panel rather than panel settings since it might be something you want to toggle as you use the panel. Changing in panel settings is a more cumbersome option.

We could also look at a right-click context menu for this setting. Right now we do not do a lot with right-click menus on panels and they raise questions of discoverability.

rgov, 2021-08-31:
As I understand it:

1. All timestamp-based plots start synchronized.
2. Panning or zooming a synchronized plot will also adjust the view of other synchronized plots, but
3. After completing a pan/zoom, a plot loses its synchronized status.

Here (3) violates my expectations; I would want a plot to remain synchronized after pan/zoom.

I would like to hold a modifier key while panning/zooming to desynchronize the plot from others. When desynchronized, a button should appear to reset. Note there are two meanings of "reset":

1. Reset the desynchronized view to match the synchronized views (if any exist).
2. Reset the view to the default range.

I think the existing "Reset View" button is too large and obscures too much of the plot.

defunctzombie, 2021-09-10:
This PR https://github.com/foxglove/studio/pull/1823 keeps plot panels always in-sync.

Opened an issue for the "reset view" button: https://github.com/foxglove/studio/issues/1834

We will probably need to open an issue for ways to "unlock" plots - but after thinking about it I think having plots remain sync'd aligns more with expectations.

PR #1823 (2021-09-11): Feature: timestamp-based plot panels remain in-sync even after pan/zoom
**User-Facing Changes**
All timestamp-based plots remain synchronized even after manual pan/zoom.

**Description**
All Timestamp-based plots start our x-axis synchronized with other timestamp-based plots. As a user pan/zooms one plot, others pan/zoom to the same x-axis window. This facilitates zooming in to anomalous events on one plot and having other plots update to the same window without the user having to manually zoom each one.

The behavior prior to this change would "unlock" a plot from others after the user panned [... cut]
commit 0ed4c052 (2021-09-11): Feature: timestamp-based plot panels remain in-sync even after pan/zoom (#1823)

All Timestamp-based plots start our x-axis synchronized with other timestamp-based plots. As a user pan/zooms one plot, others pan/zoom to the same x-axis window. This facilitates zooming in to anomalous events on one plot and having other plots update to the same window without the user having to manually zoom each one.

The behavior prior to this change would "unlock" a plot from others after the user panned/zoomed that plot. This behavior is unexpected since panning/zooming that plot would continue to pan/zoom  [... cut]
closed 2021-09-11 by defunctzombie