foxglove/studio #1738: 3d panel has precision issues when robot is very far from root transform

amacneil, 2021-08-31:
**Description**
The 3d panel gets hilariously messed up if the robot is very far (> 1000 km) from the root transform, likely due to floating point math.

- Version: v0.16
- OS: Linux

**Steps To Reproduce**

Example bag: [bug.bag.zip](https://github.com/foxglove/studio/files/7087164/bug.bag.zip)
Created using [this script](https://gist.github.com/amacneil/68e7d7094d0ed1da879dd8a6d4431720).

In this bag, `base_link` is 5000 km from `map`. The marker moves away from `base_link` at 1m/s.
Marker renders pretty well when `base_link` is under 100km from `map`, but starts getting super crazy above 5000 km, and is unviewable above 10,000km.

**Expected Behavior**
Nice buttery-smooth animation of the marker moving across the screen.

**Actual Behavior**

https://user-images.githubusercontent.com/637671/131586680-b889ff1b-7035-4f17-ad80-7b3533850dd9.mp4

jtbandes, 2021-08-31:
Some relevant code:

https://github.com/foxglove/studio/blob/050387e915aa546f04a0f82ae7258addc677e36f/packages/studio-base/src/panels/ThreeDimensionalViz/TopicTree/Layout.tsx#L439-L445

https://github.com/foxglove/studio/blob/050387e915aa546f04a0f82ae7258addc677e36f/packages/studio-base/src/panels/ThreeDimensionalViz/SceneBuilder/index.ts#L478-L483

amacneil, 2021-09-01:
It's worth mentioning that (as far as I understand) we probably can't avoid precision issues if we have objects to render which are very far from their frame, or two objects in different frames when both frames are very far from their parent. This is due to float32 only supporting ~7 significant digits. 

However, the bug here is that if we are rendering objects very close to their own frame, we shouldn't have precision issues if we are not rendering anything based on the parent frame. We should just ignore the parent frame.

ShepelIlya, 2021-09-08:
Intrested in topic too.

amacneil, 2021-09-09:
Also noting that the best workaround for now is simply to use a “site” frame, and delete any transforms to a “map” or “world” frame from your bag. Foxglove currently always uses the topmost (root) transform in the bag for calculations.

amacneil, 2021-11-19:
A short term solution might be to detect when we are rendering any objects very far from the root frame, and display a warning linking to this issue. That would at least explain to people what is happening and how to work around it.

jhurliman, 2021-12-08:
This is fixed in https://github.com/foxglove/studio/pull/2319

https://user-images.githubusercontent.com/195374/145276757-fde87ae6-fc10-4cc5-8930-11ec4645d3c4.mov

closed 2021-12-08 by jhurliman