foxglove/studio #1740: Long bags produce an unhelpful stack trace

amacneil, 2021-09-01:
**Description**
Bags longer than about 4 months produce a mildly helpful error, with an unhelpful stack trace.

- Version: 0.16
- OS: Linux

**Steps To Reproduce**
Create a bag with two messages in it, 4 months apart: [bug.bag.zip](https://github.com/foxglove/studio/files/7087380/bug.bag.zip)

```
import rosbag
from datetime import datetime
from visualization_msgs.msg import Marker

bag = rosbag.Bag('bug.bag', 'w')

try:        
    m1 = Marker()
    m1.header.stamp.secs = round(datetime(2020, 1, 1).timestamp())
    bag.write('/marker', m1, m1.header.stamp)
    
    m2 = Marker()
    m2.header.stamp.secs = round(datetime(2020, 5, 1).timestamp())
    bag.write('/marker', m2, m2.header.stamp)
    
    print('success')
finally:
    bag.close()
```

**Expected Behavior**
Helpful error message, maybe even telling me how long my bag is.

**Actual Behavior**
Stack trace:
```
Error: Time range is too long to be supported
    at en.initialize (file:///opt/Foxglove%20Studio/resources/app.asar/renderer/main.js:4187:12598)
    at async Cy.initialize (file:///opt/Foxglove%20Studio/resources/app.asar/renderer/main.js:4182:25091)
    at async St.initialize (file:///opt/Foxglove%20Studio/resources/app.asar/renderer/main.js:4188:2827)
    at async Cy.initialize (file:///opt/Foxglove%20Studio/resources/app.asar/renderer/main.js:4182:25091)
```

[image]

jtbandes, 2021-09-01:
https://github.com/foxglove/studio/blob/5697253ae962736f9b83d60255afb2c9871c58b4/packages/studio-base/src/randomAccessDataProviders/MemoryCacheDataProvider.ts#L334-L336

jhurliman, 2022-01-07:
Fixed in the latest release. Instead of a stack trace, this is now a data source error:

[image]

closed 2022-01-07 by jhurliman