foxglove/studio #1743: Team layouts UI polish (local layouts)

jtbandes, 2021-09-01:


amacneil, 2021-09-01:
dot for dirty state:

[image]

my vote is to remove the "three dots" menu entirely and rely on right-click for context menu like every other app does

jtbandes, 2021-09-01:
Right click is rarely if ever the _only_ way to access some action. It is a nice to have for people who remember to use it but I don’t think we should lean on it.

amacneil, 2021-09-01:
Fair point - although typically the same actions are available via the traditional menubar as well (not via inline icons).

E.g. VSCode and Figma both follow this pattern. Most actions are available via right click menu, and also via the menubar. Both apps also expose their app menus under a single button in browser mode (i.e. github.dev for VSCode).

I'd encourage us to follow this pattern and act more like a native app, and expose the actions via right click and a `Layout` menu or something.

Our menubar is hidden by default on Linux+Windows, but can be easily accessed by hitting `Alt` (which most power users know about on win/mac). I would be in favor of showing it all the time though (VSCode shows menubar all the time, Figma desktop seems to rely on their same monomenu button as they use in browser).

amacneil, 2021-09-01:
I think "traditional" is changing over time. Both Gnome and Windows seem to be moving away from menu bars and dumping all menus under a hamburger menu somewhere, or removing it entirely. But power user tools tend to keep it.

But for some specific examples - on Linux, both Slack and VS Code show menu bar by default, but let you configure it. In Slack you can hide it via `Window`->`Always Show Menu Bar`. In VSCode you can hide it via `View`->`Appearance`->`Show Menu Bar`. You can always access it via `Alt` shortcut.

amacneil, 2021-09-01:
I don't use Windows as frequently, but I don't think the ribbon thing is very common outside of Office and Windows Explorer?

closed 2021-09-03 by jtbandes