foxglove/studio #1749: TFs are Unsupported / Unavailable

amacneil, 2021-09-02:
**Description**
Not 100% sure this is a bug.

In this bag, my transforms work fine (I can render the scene and change viewport transform), but I can't render the transforms in the 3d scene (they show as `Unsupported`):

[image]

- Version: v0.16
- OS: linux

**Steps To Reproduce**
[nuscenes.bag.zip](https://github.com/foxglove/studio/files/7100562/nuscenes.bag.zip)

amacneil, 2021-09-02:
Solved by publishing [tf2_msgs/TFMessage](https://docs.ros.org/en/jade/api/tf2_msgs/html/msg/TFMessage.html) instead of [geometry_msgs/TransformStamped](http://docs.ros.org/en/api/geometry_msgs/html/msg/TransformStamped.html) (thanks @jhurliman).

Leaving this bug open to discuss whether we should support `geometry_msgs/TransformStamped` directly.

jhurliman, 2022-01-07:
Working in the latest release

closed 2022-01-07 by jhurliman