foxglove/studio #1760: Web sign in flow

jtbandes, 2021-09-03:
Clicking "sign in" on the web build currently uses the same device code flow as the desktop app. It should instead use an in-browser flow where the user can take advantage of existing login cookies without having to confirm a code.

dayjaby, 2021-09-16:
Is the login working at all? I only get a `Failed to fetch: https://api.foxglove.dev/v1/auth/device-code`

amacneil, 2021-09-16:
This feature hasn't been released yet, so it's not expected to work. Stay tuned!

If you want to join the beta, please DM me in our [Community Slack](https://foxglove.dev/slack) or email adrian@foxglove.dev.

defunctzombie, 2021-09-16:
@amacneil assigning this to you with the action item of scheduling a meeting/jam session to plan out the studio/console aspects here.

amacneil, 2021-09-22:
Discussed. Next steps are:

1. Fix CORS for api.foxglove.dev
2. Enable cookie authentication in Studio web (this ticket)

commit f3fb2c86 (2021-10-07): Feature: use in-browser signin flow for studio web (#1947)

**User-Facing Changes**
None

**Description**
Change the CurrentUserContext to expose a signIn function instead of bearer token management. This allows different CurrentUserContext providers to handle signin specific to their environment.

On desktop, we perform signin via the device code flow and store credentials in localstorage. This flow is moved into the ConsoleApiCurrentUserProvider.

On web, we perform signin via console and share the same credentials and session. This is implemented in the ConsoleApiCookieUserProvider.

Fixes: [... cut]