foxglove/studio #1783: Configurable TF Markers

ShepelIlya, 2021-09-06:
Is there any way to make TF markers on 3D scene widget configurable? In RViz we can set custom length and diameter of TF axes, and i think that we need similar feature in Foxglove.

amacneil, 2021-09-08:
It's not possible today (as far as I know), but I also needed this feature recently and agree we should add it.

ShepelIlya, 2021-09-08:
@amacneil There is an other way, i think. Is it possible to add nav_msgs/Odometry visualizer, that can display robot's pose and orientation? It will be much easier.

amacneil, 2021-09-08:
Yes, we should do that too. 

See https://github.com/foxglove/studio/issues/1629

clementchef, 2022-05-19:
Another use case would be to show the connection between frame (currently yellow edge) and allow to hide the axis markers. In the case of arm robots you would just see the arm skeleton moving without all the clutter created by the axis markers. The axis are useful in some case on single frame for debugging but are not great for general dynamic visualization.

jhurliman, 2022-07-14:
New3D frame axes (TF markers) have configurable display options